qosmio/openwrt-ipq #17: You can support additional router devices jdc ax1800 and jdc ax6600

Doanduy09, 2024-11-01:
Can you please support more router devices jdc ax1800 and jdc ax6600, thanks

wuruxu, 2024-11-01:
I also use jdc ax6600 too

qosmio, 2024-11-04:
I mainly add NSS support for existing devices in upstream OpenWrt. The only exception is the MX4300, which isn’t upstream yet; I currently own this device and can personally verify that everything works with each update I push.

Are there branches that support these devices? Minus NSS of course.

wuruxu, 2024-11-04:
NSS support is very great , these two devices isn't supported in upstream openwrt yet.

Doanduy09, 2024-11-05:
> Tôi chủ yếu thêm hỗ trợ NSS cho các thiết bị hiện có trong OpenWrt thượng nguồn. Ngoại lệ duy nhất là MX4300, hiện vẫn chưa được thượng nguồn; hiện tại tôi sở hữu thiết bị này và có thể tự mình xác minh rằng mọi thứ đều hoạt động với mỗi bản cập nhật tôi đưa ra.
> 
> Có nhánh nào hỗ trợ các thiết bị này không? Tất nhiên là trừ NSS.

Currently I see this branch supports jdc ax 1800 and jdc ax6600, but when I build the rom to test, I see that it doesn't support wifi very well.
https://github.com/LiBwrt-op/openwrt-6.x
https://github.com/LiBwrt-op/openwrt-6.x/tree/main/target/linux/qualcommax/files/arch/arm64/boot/dts/qcom

breeze303, 2024-11-05:
> > 我主要为上游 OpenWrt 中的现有设备添加 NSS 支持。唯一的例外是 MX4300，它尚未升级;我目前拥有这台设备，可以自己验证我发布的每个更新是否一切正常。
> > 是否有支持这些设备的分支？当然，NSS 除外。
> 
> 目前我看到这个分支支持 jdc ax 1800 和 jdc ax6600，但是当我构建 rom 进行测试时，我看到它不是很支持 wifi。 https://github.com/LiBwrt-op/openwrt-6.x https://github.com/LiBwrt-op/openwrt-6.x/tree/main/target/linux/qualcommax/files/arch/arm64/boot/dts/qcom

You can try the firmware here，Both devices have been using this firmware for close to half a year now, and there are no major problems with it so far
https://github.com/breeze303/openwrt-ci/releases/tag/IPQ60XX-ALL

Doanduy09, 2024-11-05:
> You can try the firmware here，Both devices have been using this firmware for close to half a year now, and there are no major problems with it so far https://github.com/breeze303/openwrt-ci/releases/tag/IPQ60XX-ALL

thank you very much

Doanduy09, 2024-11-05:
I built https://github.com/LiBwrt-op/openwrt-6.x compare https://github.com/VIKINGYFY/immortalwrt , same position then see that jdc AX6600 of LiBwrt is slower than VIKINGYFY, I tried many times.. the result is still the same.

qosmio, 2025-04-25:
@breeze303 I see you've committed a few changes for jdcloud ax1800/ax6600 in your fork, have you opened any PR to have the work committed upstream?

breeze303, 2025-04-29:
> [@breeze303](https://github.com/breeze303) I see you've committed a few changes for jdcloud ax1800/ax6600 in your fork, have you opened any PR to have the work committed upstream?

Some of my changes may not pass upstream scrutiny, so I haven't opened any PR.