qosmio/openwrt-ipq #21: NSS 24.10 or NSS main image for MX4300 does not work with STA/CLIENT mode for WIFI interfaces.

vochong, 2024-12-14:
### Describe the bug

With NSS 24.10 or NSS main image for MX4300, if I configure STA/Client mode on a WIFI interface (e.g. phy2-ap0), dmesg clearly shows that it's associated with the upstream WIFI AP, but it can never acquire any DHCP IP address from it. 

With FOSS 24.10 or FOSS main image for MX4300, STA/Client mode can always acquire a DHCP IP address from the upstream WIFI AP.

### OpenWrt version

24.10

### OpenWrt release

24.10

### OpenWrt target/subtarget

ipq807x

### Device

Linksys_MX4300

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vochong, 2024-12-14:
@qosmio 

STA/Client mode also has the same problem in qualcommax-6.x-nss-mx4300-6.9.

ascendbeing, 2024-12-24:
Works for me. However I am using WDS via phy1 (2.4ghz) between two mx4300s. I didn't try with phy2. I would suggest trying to enable WDS on both sides. Maybe take a look at qos's example configs. If you want I can provide a mostly working (other than spam on WDS client about seeing the same IP traffic) config snippet or snippets.

vochong, 2024-12-26:
According to a user from https://forum.openwrt.org/t/qualcommax-nss-build/148529/4290,  he mentioned that STA/Client mode worked with an NSS image (for another non-MX4300 model) dated 2024-08-26.   But STA/client mode failed in later builds, so he experienced the same issue as I did.

```
For me STA/client mode is working on my ax3600 running an old AgustinLorenzo prebuilt (NSS-WiFi) dated 2024-08-26-2004.   A few releases later the 2024-09-14-1545 build already showed the problem, i've not tested the inbetween releases.
```

vochong, 2024-12-26:
> Works for me. However I am using WDS via phy1 (2.4ghz) between two mx4300s. I didn't try with phy2. I would suggest trying to enable WDS on both sides. Maybe take a look at qos's example configs. If you want I can provide a mostly working (other than spam on WDS client about seeing the same IP traffic) config snippet or snippets. edit and yes with the default NSS branch no 6.9. worth noting I heavily customized my build in make menuconfig. Maybe related to why it works for me.

STA/Client mode and WDS mode are totally different. While STA/Client mode focuses on connecting a single device to an AP, WDS mode focuses on extending the network by linking APs. Comparing them directly is not appropriate because they address different networking needs.

ascendbeing, 2024-12-26:
did you try relayd? luci-proto-relay and it's dependencies. it may or may not help you with getting connected without wds

vochong, 2024-12-31:
We are here to report real bugs only, not to try unrelated workarounds.

qosmio, 2025-01-02:
hi, I hadn't actually come across this scenario in my testing, most of my efforts on the MX4300 have been to ensure meshing and WDS work in static/dhcp mode. Which version of NSS FW are you using? 11.4 or 12.1/12.2/12.5?

vochong, 2025-01-02:
Happy New Year to you and your family.

I used the default config in your config-nss.seed (plus a bunch of other packages), so the NSS FW is 11.4 (CONFIG_NSS_FIRMWARE_VERSION_11_4=y in .config).

qosmio, 2025-01-02:
Happy New Year!

I believe it's very likely related to `tx-checksumming`. It was causing DHCP issues for clients connected via mesh on NSS builds, so I had created a hotplug to disable it explicitly for said interfaces.

Can you try changing the following lines?

```diff
--- /etc/hotplug.d/net/99-disable_mesh_chksum_ath11k
+++ /etc/hotplug.d/net/99-disable_mesh_chksum_ath11k
@@ -11,7 +11,7 @@
   *ath11k*)
     mode=$(ubus call network.wireless status \
       | jsonfilter -e "@[*].interfaces[@.ifname='$INTERFACE'].config.mode" 2> /dev/null)
-    [ "$mode" = "mesh" ] || exit 0
+    { [ "$mode" = "mesh" ] || [ "$mode" = "sta" ]; } || exit 0

     INC="/lib/netifd/offload/disable_offloads.sh"
     . "$INC" 2> /dev/null || {
```

Then restart the interface on client device:

```sh
wifi up radio2
```

vochong, 2025-01-04:
Hello,
Disabling tx checksumming does not help with STA mode. The phyX-sta0 interface still can not acquire any IP address from the upstream AP.   I tried phy0-sta0, phy1-sta0, phy2-sta0 to no avail.

```
[  677.713421] phy2-sta0: authenticate with xx:xx:xx:xx:xx:xx(local address=80:69:1a:22:97:28)
[  677.713480] phy2-sta0: send auth to xx:xx:xx:xx:xx:xx (try 1/3)
[  677.723476] phy2-sta0: authenticated
[  677.733151] phy2-sta0: associate with xx:xx:xx:xx:xx:xx (try 1/3)
[  677.735631] phy2-sta0: RX AssocResp from xx:xx:xx:xx:xx:xx  (capab=0x11 status=0 aid=4)
[  677.739354] phy2-sta0: associated
```

```
Features for phyX-sta0:
rx-checksumming: off [fixed]
tx-checksumming: off
	tx-checksum-ipv4: off [fixed]
	tx-checksum-ip-generic: off
	tx-checksum-ipv6: off [fixed]
	tx-checksum-fcoe-crc: off [fixed]
	tx-checksum-sctp: off [fixed]
scatter-gather: off
	tx-scatter-gather: off [fixed]
	tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
	tx-tcp-segmentation: off [fixed]
	tx-tcp-ecn-segmentation: off [fixed]
	tx-tcp-mangleid-segmentation: off [fixed]
	tx-tcp6-segmentation: off [fixed]
generic-segmentation-offload: off [requested on]
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: off [fixed]
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
tx-udp_tnl-csum-segmentation: off [fixed]
tx-gso-partial: off [fixed]
tx-tunnel-remcsum-segmentation: off [fixed]
tx-sctp-segmentation: off [fixed]
tx-esp-segmentation: off [fixed]
tx-udp-segmentation: off [fixed]
tx-gso-list: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
hw-tc-offload: off [fixed]
esp-hw-offload: off [fixed]
esp-tx-csum-hw-offload: off [fixed]
rx-udp_tunnel-port-offload: off [fixed]
tls-hw-tx-offload: off [fixed]
tls-hw-rx-offload: off [fixed]
rx-gro-hw: off [fixed]
tls-hw-record: off [fixed]
rx-gro-list: off
macsec-hw-offload: off [fixed]
rx-udp-gro-forwarding: off
hsr-tag-ins-offload: off [fixed]
hsr-tag-rm-offload: off [fixed]
hsr-fwd-offload: off [fixed]
hsr-dup-offload: off [fixed]
```

vochong, 2025-01-04:
Same problem in both NSS firmware 11.4 and 12.2.

vochong, 2025-01-04:
The problem exists on other platforms as well.

https://forum.openwrt.org/t/qualcommax-nss-build/148529/4290

```
For me STA/client mode is working on my ax3600 running an old AgustinLorenzo prebuilt (NSS-WiFi) dated 2024-08-26-2004

A few releases later the 2024-09-14-1545 build already showed the problem, i've not tested the in between releases.
```

Herrnobiz, 2025-01-07:
I'm the xiaomi ax3600 user from the forum using an old release. Tried rebuilding using current nss patches and choosing a different nss firmware ver, from 12.5 to 11.4 (as someone said the older one could be working in STA) without success.
FOSS build work

vochong, 2025-01-24:
Hi @qosmio, 

Do you plan to look into this issue some time in the near future?  It seems some NSS patch changes broke it after August 2024.

Thanks!

qosmio, 2025-02-21:
I think I may have narrowed it down to `hostapd` issue. Can you test and rebuild on main-nss branch with this?
```
rm -fr package/network/services/hostapd
git checkout da11a1e20c34381891322946a3a654d8eb3f279b package/network/services/hostapd
```

Herrnobiz, 2025-02-22:
Reporting back after using the above hostapd branch doesn't work on fresh config: logs shows a bunch of command failed for hostapd and wpa_supplicant 
` 20:30:37 2025 daemon.notice netifd: radio1 (5317): Command failed: ubus call hostapd config_set { "phy": "phy1", "radio": -1, "config": "", "prev_config": "/var/run/hostapd-phy1.conf" } (Invalid argument)
then similar error on wpa_supplicant
daemon.notice netifd: radio1 (5317): Command failed: ubus call wpa_supplicant config_set { "phy": "phy1", "radio": -1, "config": [] } (Invalid argument)
and then
daemon.notice netifd: radio1 (5317): Interface  setup failed: WPA_SUPPLICANT_FAILED

qosmio, 2025-02-22:
Downgrading is a bit tricky since a lot of the include scripts are tied to a particular ucode/netifd. However you can selectively downgrade the Makefile, patches, and files/src. I did the same on my my build, and can confirm STA/AP, STA/WPA (WDS) are working on 2G and 5G and able to make and send DHCP requests.

vochong, 2025-02-22:
@qosmio 

I tried your steps with the latest main-nss but encountered these errors:

main-nss# rm -fr package/network/services/hostapd

main-nss# git reset package/network/services/hostapd
fatal: ambiguous argument 'package/network/services/hostapd': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'

main-nss# git checkout origin/main-nss package/network/services/hostapd
Updated 90 paths from 8f88255b05

main-nss# git checkout da11a1e package/network/services/hostapd/{Makefile,patches,files/src}
error: pathspec 'package/network/services/hostapd/files/src' did not match any file(s) known to git

qosmio, 2025-02-23:
Sorry about that, I was on mobile:
```sh
rm -fr package/network/services/hostapd
git reset -- package/network/services/hostapd
git checkout main-nss package/network/services/hostapd
git checkout da11a1e package/network/services/hostapd/{Makefile,patches,src}
```

vochong, 2025-02-23:
@qosmio 

I tried your steps with both main-nss and 24.10-nss branches, but the STA/Client functionality still did not work.

vochong, 2025-02-26:
@qosmio

Is it possible for you to upload your MX4300 image (with the changes you mentioned) to https://send.vis.ee   so I can try to see if works with STA/client mode on my device?   I followed the exact instructions you mentioned on both main-nss and 24.10-nss (with the git checkout 24.10-nss ... change) but it still did not work.

map-b, 2025-03-06:
On an Redmi AX6, I tried Agustin's last build with SME revert (only in main-nss  branch) and it works but it not get IP in some cases (I tried "ifup wwan" after connection success but nothing changed). I have two different AP to connect with, one is Xiaomi R3G with stock firmware (In this case is not fully working), the other one is a Mitrastar with stock firmware (in this case worked), both using same encryption WPA2 in the 5Ghz spectrum. 

My observations are that, first, some changes in the netifd caused to not get DHCP IP form AP until an ifup on wwan interface was made (August 2024). And later on, the SME authentication problem appeared(probably due to that hostapd bump). (Last build without SME error was on August 14, based on Augustin's builds)

I haven't bisected when this got fixed in the non-NSS openwrt but last test I did it was working ok.

vochong, 2025-03-15:
I only use WPA2 (which does not use SME authentication) and STA/CLIENT mode does not work with NSS builds.  STA/CLIENT mode in non-NSS builds always works for me.

m0eak, 2025-03-15:
Same problem.

Herrnobiz, 2025-03-30:
Thank you very much Qosmio, STA finally works again on my ax3600 on current build after removing just those 2 patches as per your instructions:
$ rm package/kernel/mac80211/patches/nss/ath11k/999-902-ath11k-fix-WDS-by-disabling-nwds.patch
$ rm package/kernel/mac80211/patches/nss/subsys/{999-775-wifi-mac80211-Changes-for-WDS-MLD.patch,999-922-mac80211-fix-null-chanctx-warning-for-NSS-dynamic-VLAN.patch}
built on main branch, all on default config (just tweaked to 512mb ram as my device)

vochong, 2025-03-30:
@Herrnobiz 

Where did qosmio ask you to remove these 2 patches to make STA work?   I did not such mention anywhere.

Removing these patches mean WDS functionality is no longer functional?

Herrnobiz, 2025-03-30:
@vochong 
Qosmio replied to the user junky in the general openwrt ipq nss forum about this solution
https://forum.openwrt.org/t/qualcommax-nss-build/148529/4474

vochong, 2025-03-30:
Hi @qosmio 

Herrnobiz confirmed that the removal of the patches makes STA mode work again.  Will you plan to fix this issue in some new commits?   Thanks a lot.

qosmio, 2025-04-25:
Forgot to mark this as completed.

For reference, the fix was applied to both `24.10-nss` and `main-nss` branches:

21c823b7 (main-nss) - ath11k_nss: Fix WDS failing to initialize or aquire IP
d2b1b37f (24.10-nss) - ath11k_nss: Fix WDS failing to initialize or aquire IP

Please be sure to use NSS FW 11.4 in your entire setup (AP and STA). Anything higher will not work.

closed 2025-04-25 by qosmio