qosmio/openwrt-ipq #24: ath11k patch failure stops build in it's tracks

ascendbeing, 2024-12-27:
### Describe the bug

069-ath11k-add-HE... fails go apply. hunk 4, 1/6 falling.

I'll try to update this report later but I'm for now chronic long this is an actual occurring issue, at least for me.

### OpenWrt version

Yours

### OpenWrt release

Yours

### OpenWrt target/subtarget

mx4300

### Device

mx4300

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ascendbeing, 2024-12-27:
For anyone who's following this, use the 24.xx branches
the main one is on an older kernel if I'm not mistaken. main is I think on 63 vs 67 on the 2xx, which built fine

sppmasterspp, 2024-12-27:
> For anyone who's following this, use the 24.xx branches the main one is on an older kernel if I'm not mistaken. main is I think on 63 vs 67 on the 2xx, which built fine

Main branch is on 6.6.67 to but building fails with

```
Applying /home/debian/openwrt-ipq-QNAP/package/kernel/mac80211/patches/nss/ath11k/069-ath11k-add-HE-stats-in-peer-stats.patch using plaintext: 
patching file drivers/net/wireless/ath/ath11k/core.h
Hunk #2 succeeded at 519 (offset 6 lines).
Hunk #3 succeeded at 528 (offset 6 lines).
Hunk #4 succeeded at 647 (offset 6 lines).
patching file drivers/net/wireless/ath/ath11k/debugfs_sta.c
patching file drivers/net/wireless/ath/ath11k/dp.h
Hunk #4 FAILED at 1422.
Hunk #5 succeeded at 1412 (offset -23 lines).
Hunk #6 succeeded at 1430 (offset -23 lines).
1 out of 6 hunks FAILED -- saving rejects to file drivers/net/wireless/ath/ath11k/dp.h.rej
patching file drivers/net/wireless/ath/ath11k/dp_rx.c
patching file drivers/net/wireless/ath/ath11k/rx_desc.h
patching file drivers/net/wireless/ath/ath11k/peer.h
Patch failed!  Please fix /home/debian/openwrt-ipq-QNAP/package/kernel/mac80211/patches/nss/ath11k/069-ath11k-add-HE-stats-in-peer-stats.patch!
make[3]: *** [Makefile:481: /home/debian/openwrt-ipq-QNAP/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/mac80211-regular/backports-6.12.6/.prepared_7686daf8eb00761a8a140532131424f0_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/debian/openwrt-ipq-QNAP/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.25#1.88#2.80
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:185: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/debian/openwrt-ipq-QNAP'
make[1]: *** [package/Makefile:179: /home/debian/openwrt-ipq-QNAP/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian/openwrt-ipq-QNAP'
make: *** [/home/debian/openwrt-ipq-QNAP/include/toplevel.mk:233: world] Error 2
```

hingbong, 2024-12-28:
same here with `main-nss`
```
2024-12-28T02:30:44.7524734Z Applying /home/runner/work/ax3600-builder/ax3600-builder/openwrt/package/kernel/mac80211/patches/nss/ath11k/069-ath11k-add-HE-stats-in-peer-stats.patch using plaintext: 
2024-12-28T02:30:44.7534872Z patching file drivers/net/wireless/ath/ath11k/core.h
2024-12-28T02:30:44.7535429Z Hunk #2 succeeded at 519 (offset 6 lines).
2024-12-28T02:30:44.7535887Z Hunk #3 succeeded at 528 (offset 6 lines).
2024-12-28T02:30:44.7536317Z Hunk #4 succeeded at 647 (offset 6 lines).
2024-12-28T02:30:44.7538806Z patching file drivers/net/wireless/ath/ath11k/debugfs_sta.c
2024-12-28T02:30:44.7542735Z patching file drivers/net/wireless/ath/ath11k/dp.h
2024-12-28T02:30:44.7544855Z Hunk #4 FAILED at 1422.
2024-12-28T02:30:44.7545272Z Hunk #5 succeeded at 1412 (offset -23 lines).
2024-12-28T02:30:44.7545740Z Hunk #6 succeeded at 1430 (offset -23 lines).
2024-12-28T02:30:44.7547052Z 1 out of 6 hunks FAILED -- saving rejects to file drivers/net/wireless/ath/ath11k/dp.h.rej
2024-12-28T02:30:44.7551057Z patching file drivers/net/wireless/ath/ath11k/dp_rx.c
2024-12-28T02:30:44.7557661Z patching file drivers/net/wireless/ath/ath11k/rx_desc.h
2024-12-28T02:30:44.7559803Z patching file drivers/net/wireless/ath/ath11k/peer.h
2024-12-28T02:30:44.7562979Z Patch failed!  Please fix /home/runner/work/ax3600-builder/ax3600-builder/openwrt/package/kernel/mac80211/patches/nss/ath11k/069-ath11k-add-HE-stats-in-peer-stats.patch!
2024-12-28T02:30:44.7567327Z make[3]: *** [Makefile:481: /home/runner/work/ax3600-builder/ax3600-builder/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/mac80211-regular/backports-6.12.6/.prepared_451a822778208012e7bc5676d6cd3492_6664517399ebbbc92a37c5bb081b5c53] Error 1
```

vochong, 2024-12-29:
https://forum.openwrt.org/t/qualcommax-nss-build/148529/4296

ascendbeing, 2025-01-01:
I think it's fixed now. Haven't tried latest push, but previous from a day or two ago built fine with main-nss-mx4300

closed 2025-01-01 by ascendbeing