qosmio/openwrt-ipq #26: mac80211 errors when connecting to ath11k with dynamic vlan

tardeaux, 2025-01-01:
### Describe the bug

When attempting to connect a client with a dynamic VLAN assignment to the ath11k radio, multiple warnings are displayed in the system log and the connection fails.  If the same config is moved to the ath10k radio in the Xiaomi AX3600, the connection is successful.  This issue is not present in the non-NSS builds.

### OpenWrt version

r28616-2f79e64ebc

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600 / Linksys LN1301

### Image kind

Self-built image

### Steps to reproduce

Configure dynamic vlan:

`root@OpenWrt-10:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd1f:1574:f95d::/48'

config interface 'lan'
        option device 'br-vlan1'
        option proto 'none'
        option defaultroute '0'

config interface 'mgmt'
        option proto 'static'
        option device 'br-vlan10'
        option ipaddr '192.168.10.10'
        option netmask '255.255.255.0'
        option gateway '192.168.10.1'
        list dns '192.168.10.1'

config device
        option type 'bridge'
        option name 'br-vlan1'
        option igmp_snooping '1'
        list ports 'lan1'
        list ports 'lan3'
        list ports 'wan'

config device
        option type 'bridge'
        option name 'br-vlan10'
        option igmp_snooping '1'
        list ports 'wan.10'

config interface 'kids'
        option proto 'none'
        option device 'br-vlan20'
        option defaultroute '0'
[... 537 lines cut ...]
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_rrdtool1=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_watchcat=y
CONFIG_PACKAGE_wpa-supplicant=y
CONFIG_PACKAGE_wpad-basic-mbedtls=m
CONFIG_TARGET_OPTIMIZATION="-O2 -pipe -mcpu=cortex-a53+crc+crypto"
CONFIG_TARGET_OPTIONS=y
CONFIG_USE_GC_SECTIONS=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

AgustinLorenzo, 2025-01-19:
Hi @tardeaux,

Did you see this? https://github.com/openwrt/openwrt/pull/17644

Regards, Agustín

tardeaux, 2025-01-20:
> Hi [@tardeaux](https://github.com/tardeaux),
> 
> Did you see this? [openwrt#17644](https://github.com/openwrt/openwrt/pull/17644)
> 
> Regards, Agustín

This is actually the exact patch that I've been using successfully on my vanilla APs.

I notice that the one included in the nss/ath11k patches folder in the NSS forks is actually the same author and is about a year and a half newer.

I actually have not tried replacing the bundled patch with this older version which is working on my non-nss devices.  I'll give it a shot and report back.

AgustinLorenzo, 2025-01-20:
Hi @tardeaux,

Yeah, this patch is very similar to that: https://github.com/qosmio/openwrt-ipq/blob/main-nss/package/kernel/mac80211/patches/nss/ath11k/207-ath11k-Add-support-for-dynamic-vlan.patch to apply OpenWRT you will first have to delete this one.

Perhaps, this is an updated version, although from the dates it does not seem so.

FYI @qosmio 

Regards, Agustin

tardeaux, 2025-01-22:
Unfortunately, if the patches are simply replaced, 244-ath11k-dp-tx-perf.patch has multiple patching failures.
There may be others further down the line, but that's where the build fails.
I don't have the knowledge to correct this.

qosmio, 2025-03-01:
@tardeaux please test with latest commits:

`main-nss`
[ath11k_nss: fix null chanctx warning for NSS dynamic VLAN
](https://github.com/qosmio/openwrt-ipq/commit/2525d6ee66291b0bcba79253708dae6161a0d23f)

`24.10-nss`
[ath11k_nss: fix null chanctx warning for NSS dynamic VLAN
](https://github.com/qosmio/openwrt-ipq/commit/323025844fb58fa8669268bcaa976b6e52ad2c37)

tardeaux, 2025-03-03:
I tried the main-nss branch (as I'm presently running my APs on the main vanilla branch code successfully)
Most of the spam seems to be gone, however there does not seem to be any traffic passed after VLAN assignment.

When I do connect to a VLAN other than 1 (which is configured as untagged traffic currently), I do get a trace on the first attempt only.  Subsequent retries do not give any warnings, but also don't pass the DHCP request.
--------------------

Sun Mar  2 17:54:11 2025 daemon.info hostapd: wlan0: STA d4:3a:2c:62:f1:e7 RADIUS: VLAN ID 40
Sun Mar  2 17:54:11 2025 daemon.info hostapd: wlan0: STA d4:3a:2c:62:f1:e7 IEEE 802.11: authenticated
Sun Mar  2 17:54:11 2025 daemon.info hostapd: wlan0: STA d4:3a:2c:62:f1:e7 IEEE 802.11: associated (aid 1)
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.038063] ------------[ cut here ]------------
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.038097] wlan0.40: Failed check-sdata-in-driver check, flags: 0x1
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.041814] WARNING: CPU: 2 PID: 2299 at ../mac80211-smallbuffers/backports-6.12.6/net/mac80211/driver-ops.c:143 drv_sta_state+0xd8/0x334 [mac80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.048102] Modules linked in: ecm(O) pppoe ppp_async nft_fib_inet nf_flow_table_inet l2tp_ppp ath11k_ahb(O) ath11k(O) pptp pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) slhc qrtr_smd qrtr qmi_helpers(O) nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c compat(O) qca_nss_wifi_meshmgr(O) qca_nss_bridge_mgr(O) qca_nss_vlan(O) qca_mcs(O) ip_gre gre qca_nss_drv(O) l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip_tunnel sha512_generic sha512_arm64 seqiv sha3_generic jitterentropy_rng drbg michael_mic hmac geniv cmac leds_pca963x leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom qca_nss_dp(O) qca_ssdk(O) gpio_button_hotplug(O) ext4 mbcache jbd2 aquantia hwmon crc_ccitt crc32c_generic
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.126373] CPU: 2 PID: 2299 Comm: hostapd Tainted: G        W  O       6.6.80 #0
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.148611] Hardware name: Linksys MX4300 (DT)
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.155982] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.160326] pc : drv_sta_state+0xd8/0x334 [mac80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.167181] lr : drv_sta_state+0xd8/0x334 [mac80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.172390] sp : ffffffc08596b640
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.177333] x29: ffffffc08596b640 x28: ffffff800318cf80 x27: ffffffc08596bdc8
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.180553] x26: ffffff8018b68900 x25: ffffff8018b69d00 x24: ffffff8009bf08c0
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.187672] x23: ffffff801ca2e000 x22: 0000000000000003 x21: 0000000000000004
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.194790] x20: ffffff8009bf08c0 x19: ffffff8018b68900 x18: 0000000000000211
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.201907] x17: 0000000000000000 x16: 0000000000000000 x15: ffffffc080d0c1e0
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.209024] x14: 0000000000000633 x13: 0000000000000211 x12: 00000000ffffffea
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.216144] x11: 00000000ffffefff x10: ffffffc080d641e0 x9 : ffffffc080d0c188
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.223261] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000001
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.230379] x5 : ffffff807fbb8850 x4 : 0000000000000000 x3 : 0000000000000027
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.237498] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000038
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.244617] Call trace:
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.251725]  drv_sta_state+0xd8/0x334 [mac80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.253985]  ieee80211_find_sta_by_link_addrs+0x6dc/0x7f4 [mac80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.258851]  sta_info_move_state+0x14/0x20 [mac80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.265184]  ieee80211_nan_func_terminated+0xd0c/0x482c [mac80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.270135]  ieee80211_nan_func_terminated+0x22b4/0x482c [mac80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.276211]  ieee80211_channel_switch+0x1e8/0x888 [mac80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.282374]  cfg80211_vendor_cmd_reply+0x1e78/0x57b4 [cfg80211]
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.288276]  genl_family_rcv_msg_doit+0xb0/0x118
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.293917]  genl_rcv_msg+0x1b4/0x24c
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.298775]  netlink_rcv_skb+0x60/0x130
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.302334]  genl_rcv+0x38/0x50
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.305978]  netlink_unicast+0x1e8/0x2d0
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.309105]  netlink_sendmsg+0x19c/0x3d0
Sun Mar  2 17:54:11 2025 kern.warn kernel: [  628.313273]  ____sys_sendmsg+0x1cc/0x27c
[... cut ...]
Sun Mar  2 17:54:11 2025 daemon.info hostapd: wlan0: STA d4:3a:2c:62:f1:e7 WPA: pairwise key handshake completed (RSN)
Sun Mar  2 17:54:11 2025 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED d4:3a:2c:62:f1:e7
Sun Mar  2 17:54:29 2025 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED d4:3a:2c:62:f1:e7
Sun Mar  2 17:54:29 2025 daemon.info hostapd: wlan0: STA d4:3a:2c:62:f1:e7 IEEE 802.11: disassociated
Sun Mar  2 17:54:30 2025 daemon.info hostapd: wlan0: STA d4:3a:2c:62:f1:e7 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

qosmio, 2025-03-06:
I've come to realize, my previous patch [ath11k_nss: fix null chanctx warning for NSS dynamic VLAN](https://github.com/qosmio/openwrt-ipq/commit/2525d6ee66291b0bcba79253708dae6161a0d23f) is actually just masking the issue.

```node
Wed Mar  5 22:29:52 2025 kern.debug kernel: [27225.021631] ath11k c000000.wifi: <?> nss vdev set cmd success cmd:14 val:1
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.247716] ath11k c000000.wifi: <?> nss peer_create msg success mac:f2:4c:df:be:22:f2 vdev:0 peer_id:4 hw_ast_idx:763 ast_hash:763
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.247783] ath11k c000000.wifi: <?> nss wifili event received 6 response 0 error 0
Wed Mar  5 22:29:57 2025 daemon.info hostapd: phy2-ap0: STA f2:4c:df:be:22:f2 IEEE 802.11: authenticated
Wed Mar  5 22:29:57 2025 daemon.info hostapd: phy2-ap0: STA f2:4c:df:be:22:f2 IEEE 802.11: associated (aid 1)
Wed Mar  5 22:29:57 2025 daemon.notice hostapd: nl80211: phy2-ap0[97]: set_sta_vlan(f2:4c:df:be:22:f2, ifname=phy2-ap0[97], vlan_id=0)
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.251143] ath11k c000000.wifi: <?> nss vdev set cmd success cmd:14 val:1
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320779] ath11k c000000.wifi: <?> nss vdev set cmd success cmd:15 val:6
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320814] ath11k c000000.wifi: <?> nss peer id 3 security cfg complete
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320829] ath11k c000000.wifi: <?> nss peer id 3 auth cfg complete
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320951] ath11k c000000.wifi: <?> nss wifili event received 21 response 0 error 0
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320980] ath11k c000000.wifi: <?> nss wifili event received 44 response 0 error 0
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320993] ath11k c000000.wifi: <?> nss vdev set cmd success cmd:15 val:6
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.321009] ath11k c000000.wifi: <?> nss peer id 3 security cfg complete
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.321022] ath11k c000000.wifi: <?> nss peer id 3 auth cfg complete
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.321125] ath11k c000000.wifi: <?> nss wifili event received 21 response 0 error 0
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.321142] ath11k c000000.wifi: <?> nss wifili event received 44 response 0 error 0
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361040] ath11k c000000.wifi: <?> vif type 3, encap type 2
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361067] ath11k c000000.wifi: <?>
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361080] ath11k c000000.wifi: nss tx msdu: 28 00000000: f2 4c df be 22 f2 80 69 1a 22 97 78 88 8e 02 03  ..}....i.".x....
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361090] ath11k c000000.wifi: nss tx msdu: 28 00000010: 00 5f 02 00 8a 00 10 00 00 00 00 00 00 00 01 ce  ._..............
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361099] ath11k c000000.wifi: nss tx msdu: 28 00000020: e3 73 fb 77 37 25 80 02 51 8c 01 ad d4 e8 8d c6  .s.w7%..Q.......
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361109] ath11k c000000.wifi: nss tx msdu: 28 00000030: 2b 40 41 d0 d7 e0 98 be 69 58 bc 81 8b 1e b0 00  +@A.....iX......
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361118] ath11k c000000.wifi: nss tx msdu: 28 00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361127] ath11k c000000.wifi: nss tx msdu: 28 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361136] ath11k c000000.wifi: nss tx msdu: 28 00000060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361146] ath11k c000000.wifi: nss tx msdu: 28 00000070: 00                                               .
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363489] ath11k c000000.wifi: <?>
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363518] ath11k c000000.wifi: dp rx msdu from nss: 00000000: 80 69 1a 22 97 78 f2 4c df be 22 f2 88 8e 02 03  .i.".x..}.......
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363533] ath11k c000000.wifi: dp rx msdu from nss: 00000010: 00 75 02 01 0a 00 10 00 00 00 00 00 00 00 01 0e  .u..............
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363546] ath11k c000000.wifi: dp rx msdu from nss: 00000020: 93 45 a2 8b e2 2b ee ec af a6 ca 9e 90 c0 f2 37  .E...+.........7
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363559] ath11k c000000.wifi: dp rx msdu from nss: 00000030: 32 9b 50 52 fc f0 0a 71 8d 70 0d f8 40 55 9a 00  2.PR...q.p..@U..
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363571] ath11k c000000.wifi: dp rx msdu from nss: 00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363584] ath11k c000000.wifi: dp rx msdu from nss: 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 b6  ................
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363597] ath11k c000000.wifi: dp rx msdu from nss: 00000060: 26 4c 21 c6 90 7e c1 8d 1a 04 e1 6f fe 77 58 00  &L!..~.....o.wX.
Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363610] ath11k c000000.wifi: dp rx msdu from nss: 00000070: 16 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01  .0..............
[... cut ...]
Wed Mar  5 22:29:57 2025 daemon.notice hostapd: phy2-ap0: EAPOL-4WAY-HS-COMPLETED f2:4c:df:be:22:f2
```

As soon as a client associates, all sdata checks start to fail. The ext vdev vlan interface never comes up, which is why you see it always failing at `drv_sta_state+0xd8/0x34c`. This is a pretty big endeavor, and after working all this week, I'm no where closer to getting this tackled.

I will have to leave it open for now. Hopefully @BrainSlayer has come across the same issue. He was very helpful with WDS issues I had last year.

BrainSlayer, 2025-03-06:
> I've come to realize, my previous patch [ath11k_nss: fix null chanctx warning for NSS dynamic VLAN](https://github.com/qosmio/openwrt-ipq/commit/2525d6ee66291b0bcba79253708dae6161a0d23f) is actually just masking the issue.
> 
> Wed Mar  5 22:29:52 2025 kern.debug kernel: [27225.021631] ath11k c000000.wifi: <?> nss vdev set cmd success cmd:14 val:1
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.247716] ath11k c000000.wifi: <?> nss peer_create msg success mac:f2:4c:df:be:22:f2 vdev:0 peer_id:4 hw_ast_idx:763 ast_hash:763
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.247783] ath11k c000000.wifi: <?> nss wifili event received 6 response 0 error 0
> Wed Mar  5 22:29:57 2025 daemon.info hostapd: phy2-ap0: STA f2:4c:df:be:22:f2 IEEE 802.11: authenticated
> Wed Mar  5 22:29:57 2025 daemon.info hostapd: phy2-ap0: STA f2:4c:df:be:22:f2 IEEE 802.11: associated (aid 1)
> Wed Mar  5 22:29:57 2025 daemon.notice hostapd: nl80211: phy2-ap0[97]: set_sta_vlan(f2:4c:df:be:22:f2, ifname=phy2-ap0[97], vlan_id=0)
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.251143] ath11k c000000.wifi: <?> nss vdev set cmd success cmd:14 val:1
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320779] ath11k c000000.wifi: <?> nss vdev set cmd success cmd:15 val:6
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320814] ath11k c000000.wifi: <?> nss peer id 3 security cfg complete
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320829] ath11k c000000.wifi: <?> nss peer id 3 auth cfg complete
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320951] ath11k c000000.wifi: <?> nss wifili event received 21 response 0 error 0
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320980] ath11k c000000.wifi: <?> nss wifili event received 44 response 0 error 0
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.320993] ath11k c000000.wifi: <?> nss vdev set cmd success cmd:15 val:6
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.321009] ath11k c000000.wifi: <?> nss peer id 3 security cfg complete
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.321022] ath11k c000000.wifi: <?> nss peer id 3 auth cfg complete
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.321125] ath11k c000000.wifi: <?> nss wifili event received 21 response 0 error 0
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.321142] ath11k c000000.wifi: <?> nss wifili event received 44 response 0 error 0
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361040] ath11k c000000.wifi: <?> vif type 3, encap type 2
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361067] ath11k c000000.wifi: <?>
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361080] ath11k c000000.wifi: nss tx msdu: 28 00000000: f2 4c df be 22 f2 80 69 1a 22 97 78 88 8e 02 03  ..}....i.".x....
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361090] ath11k c000000.wifi: nss tx msdu: 28 00000010: 00 5f 02 00 8a 00 10 00 00 00 00 00 00 00 01 ce  ._..............
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361099] ath11k c000000.wifi: nss tx msdu: 28 00000020: e3 73 fb 77 37 25 80 02 51 8c 01 ad d4 e8 8d c6  .s.w7%..Q.......
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361109] ath11k c000000.wifi: nss tx msdu: 28 00000030: 2b 40 41 d0 d7 e0 98 be 69 58 bc 81 8b 1e b0 00  +@A.....iX......
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361118] ath11k c000000.wifi: nss tx msdu: 28 00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361127] ath11k c000000.wifi: nss tx msdu: 28 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361136] ath11k c000000.wifi: nss tx msdu: 28 00000060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.361146] ath11k c000000.wifi: nss tx msdu: 28 00000070: 00                                               .
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363489] ath11k c000000.wifi: <?>
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363518] ath11k c000000.wifi: dp rx msdu from nss: 00000000: 80 69 1a 22 97 78 f2 4c df be 22 f2 88 8e 02 03  .i.".x..}.......
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363533] ath11k c000000.wifi: dp rx msdu from nss: 00000010: 00 75 02 01 0a 00 10 00 00 00 00 00 00 00 01 0e  .u..............
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363546] ath11k c000000.wifi: dp rx msdu from nss: 00000020: 93 45 a2 8b e2 2b ee ec af a6 ca 9e 90 c0 f2 37  .E...+.........7
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363559] ath11k c000000.wifi: dp rx msdu from nss: 00000030: 32 9b 50 52 fc f0 0a 71 8d 70 0d f8 40 55 9a 00  2.PR...q.p..@U..
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363571] ath11k c000000.wifi: dp rx msdu from nss: 00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363584] ath11k c000000.wifi: dp rx msdu from nss: 00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 b6  ................
> Wed Mar  5 22:29:57 2025 kern.debug kernel: [27229.363597] ath11k c000000.wifi: dp rx msdu from nss: 00000060: 26 4c 21 c6 90 7e c1 8d 1a 04 e1 6f fe 77 58 00  &L!..~.....o.wX.
[... cut ...]
> As soon as a client associates, all sdata checks start to fail. The ext vdev vlan interface never comes up, which is why you see it always failing at `drv_sta_state+0xd8/0x34c`. This is a pretty big endeavor, and after working all this week, I'm no where closer to getting this tackled.
> 
> I will have to leave it open for now. Hopefully [@BrainSlayer](https://github.com/BrainSlayer) has come across the same issue. He was very helpful with WDS issues I had last year.

not yet. but i see "channel_switch" in the backtrace. dfs related?

tardeaux, 2025-03-06:
@qosmio I'm in no particular rush personally since my nodes that use dyn vlan are working fine on the vanilla builds and fine to be less speedy.  They cooperate just fine with my wireless NSS bridges that only serve wired clients.

I can add that I got some time to do packet captures while clients were trying to connect and I was able to observe that while the clients were being assigned a VLAN by RADIUS, all dhcp requests were going out on the untagged network and getting replies all the way up to the radio's wlan0(untagged) interface and then disappearing, never making it to the client, even on the untagged vlan1.  I'm presently working on restructuring my network move all client traffic off of vlan1 so future troubleshooting isn't impacted by UniFi's weird treatment there.  I'll make sure to include updated config files in future posts.

qosmio, 2025-03-06:
@BrainSlayer, I thought that it might be that, but I tried on non-DFS and even on 2G. Same sequence of events. NSS is doing something strange as soon as hostapd moves the STA from the main AP to the VLAN.

Decoded stack trace to code + line numbers
```c
====  drv_sta_state+0xd8/0x34c [mac80211] ====
0x4524 is in drv_sta_state (../mac80211-regular/backports-6.12.6/net/mac80211/driver-ops.c:143).
138		if (!ieee80211_hw_check(&local->hw, SUPPORTS_NSS_OFFLOAD) ||
139		    !(old_state == IEEE80211_STA_ASSOC &&
140		      new_state == IEEE80211_STA_AUTHORIZED))
141			sdata = get_bss_sdata(sdata);
142
143-->		if (!check_sdata_in_driver(sdata))
144			return -EIO;
145
146		trace_drv_sta_state(local, sdata, &sta->sta, old_state, new_state);
147		if (local->ops->sta_state) {

====  ieee80211_find_sta_by_link_addrs+0x6e0/0x7d4 [mac80211] ====
0x71d0 is in _sta_info_move_state (../mac80211-regular/backports-6.12.6/net/mac80211/sta_info.c:1333).
1328		 * fail the transition
1329		 */
1330		if (test_sta_flag(sta, WLAN_STA_INSERTED)) {
1331			int err = drv_sta_state(sta->local, sta->sdata, sta,
1332						sta->sta_state, new_state);
1333-->			if (err)
1334				return err;
1335		}
1336
1337		/* reflect the change in all state variables */

====  sta_info_move_state+0x14/0x20 [mac80211] ====
0x8780 is in sta_info_move_state (../mac80211-regular/backports-6.12.6/net/mac80211/sta_info.c:1415).
1410
1411	int sta_info_move_state(struct sta_info *sta,
1412				enum ieee80211_sta_state new_state)
1413	{
1414		return _sta_info_move_state(sta, new_state, true);
1415-->	}
1416
1417	static void __sta_info_destroy_part2(struct sta_info *sta, bool recalc)
1418	{
1419		struct ieee80211_local *local = sta->local;

====  ieee80211_nan_func_terminated+0x1288/0x4af0 [mac80211] ====
0x23ff8 is in sta_apply_auth_flags (../mac80211-regular/backports-6.12.6/net/mac80211/cfg.c:1726).
1721				ret = sta_info_move_state(sta, IEEE80211_STA_AUTHORIZED);
1722			else if (test_sta_flag(sta, WLAN_STA_AUTHORIZED))
1723				ret = sta_info_move_state(sta, IEEE80211_STA_ASSOC);
1724			else
1725				ret = 0;
1726-->			if (ret)
1727				return ret;
1728		}
1729
1730		if (mask & BIT(NL80211_STA_FLAG_ASSOCIATED) &&

====  ieee80211_nan_func_terminated+0x2728/0x4af0 [mac80211] ====
0x25498 is in sta_apply_parameters (../mac80211-regular/backports-6.12.6/net/mac80211/cfg.c:1978).
1973		/* auth flags will be set later for TDLS,
1974		 * and for unassociated stations that move to associated */
1975		if (!test_sta_flag(sta, WLAN_STA_TDLS_PEER) &&
1976		    !((mask & BIT(NL80211_STA_FLAG_ASSOCIATED)) &&
1977		      (set & BIT(NL80211_STA_FLAG_ASSOCIATED)))) {
1978-->			ret = sta_apply_auth_flags(local, sta, mask, set);
1979			if (ret)
1980				return ret;
1981		}
1982

====  ieee80211_channel_switch+0x4ac/0x5f8 [mac80211] ====
0x27d0c is in ieee80211_change_station (../mac80211-regular/backports-6.12.6/net/mac80211/cfg.c:2251).
2246				cfg80211_send_layer2_update(sta->sdata->dev,
2247							    sta->sta.addr);
2248			}
2249		}
2250
2251-->		err = sta_apply_parameters(local, sta, params);
2252		if (err)
2253			return err;
2254
2255		if (sdata->vif.type == NL80211_IFTYPE_STATION &&

====  cfg80211_vendor_cmd_reply+0x1e58/0x5a64 [cfg80211] ====
0x1a100 is in nl80211_set_station (../mac80211-regular/backports-6.12.6/net/wireless/rdev-ops.h:223).
218					      struct net_device *dev, u8 *mac,
219					      struct station_parameters *params)
220	{
221		int ret;
222		trace_rdev_change_station(&rdev->wiphy, dev, mac, params);
223-->		ret = rdev->ops->change_station(&rdev->wiphy, dev, mac, params);
224		trace_rdev_return_int(&rdev->wiphy, ret);
225		return ret;
226	}
227
```

I experimented with removing the OFFLOAD check so that ```sdata = get_bss_sdata(sdata)``` gets unconditionally set but that just results `encap mismatch errors`.

BrainSlayer, 2025-03-07:
from this log the error does no occur in ath11k, but in mac80211 itself. check_sdata_in_driver fails. this technically means the interface cannot be added, because its already there. but i need to read alot of code here before i can tell you more and what the possible fix would be

BrainSlayer, 2025-03-07:
@qosmio  okay. its likelly that your port from nss qdsk patches are incomplete. check patch 920 and  patch 913 from patches/subsys for instance

BrainSlayer, 2025-03-07:
another curious observation is that hostapd calls "set_sta_vlan" twice. which leads to nl80211_set_station called twice

BrainSlayer, 2025-03-07:
okay. most likelly cause is that qca does not even provide patches for mac80211/ath11k with nss acceleration but there are also patches for hostapd with nss. i just found ap vlan related patches in hostapd on the qsdk

BrainSlayer, 2025-03-07:
@qosmio  please check this patch i took from qsdk https://github.com/mirror/dd-wrt/commit/d0f7474f3a9cd31332bc8e34069fbb703b10c428

qosmio, 2025-03-07:
@BrainSlayer, thank you for the tips! I actually have that patch in my testing repo too. Plus a few others I added to see it it helped.

```diff
package/network/services/hostapd/patches/h00-003-hostapd-allow-AP_VLAN-creation-for-dynamic-VLAN.patch 

package/network/services/hostapd/patches/r33-0001-QSDK-hostapd-Fix-interface-addition-to-bridge-sequen.patch

package/network/services/hostapd/patches/r34-hostapd-add-ap-vlan-interface-to-bridge-after-interface-up.patch
```

Sadly same issue.

> another curious observation is that hostapd calls "set_sta_vlan" twice. which leads to nl80211_set_station called twice

I might be wrong in my assumption, but isn't that mean to happen? When STA is associated and authorized set to VID 0 (the default behavior). After that, hostapd sends another signal to mac80211 to move the STA to newly created bridge vlan interface?

Are dynamic VLANs working with NSS in dd-wrt?

BrainSlayer, 2025-03-08:
i never played with dynamic vlans nor did i ever implement it. i mean it might work. but i cannot give you a answer here. but moving the sta to a new bridge must be handled by hotplug events. mac80211 is not removing interfaces from bridges and adding it to another one. for wds it works like its authenticated and then the interface is added and the hotplug code has to decide where to add it. i dont think this is much different from ap_vlan since they share most of the code. wds sta is handled like ap_vlan station

BrainSlayer, 2025-03-08:
from the patches you mentioned i see that hostapd is indeed handling bridge addition and removal. weired

BrainSlayer, 2025-03-08:
from my point of view it seem that things got changed in mac80211 which are not working well anymore with the nss patches here. some workflow. check_sdata_in_driver fails here because a flag is set which shouldnt be there. and we have to find out why.

pwned-pixel, 2025-03-09:
I've tested this setup a while ago before it was introduced into OpenWrt officially. But I had no luck to get it working with NSS build. Basically I had three issues:

1.) After each wifi (re-)start the vlans were not distributed to each phy-ap. It was like random. I've checked it with "brctl show".

I cannot remember what I did exactly but if I remember correct I've circumvented this issue with by creating "wifi stations" assigning a password to each vlan.

```
config wifi-vlan
        option vid '110'
        option name '110'
        option network 'lan'

config wifi-station
        option vid '110'
        option key '****'
```

2.) There were random crashes with this setup in the past. I've still a crashlog here. But I don't know if this is still relevant. Maybe it helps:

```
[  103.942829] phy1-ap0-110: Failed check-sdata-in-driver check, flags: 0x1
[  103.946543] WARNING: CPU: 1 PID: 2890 at /home/flynn/nolf/openwrt/zyxel_nbg7815/openwrt/build_dir/target-aarch64-openwrt-linux-musl_musl/linux-qualcommax_ipq807x/mac80211-regular/backports-6.11.2/net/mac80211/driver-ops.c:113 drv_remove_interface+0x70/0xc4 [mac80211]
[  103.954278] Modules linked in: ecm(O) pppoe ppp_async l2tp_ppp ath11k_pci(O) ath11k_ahb(O) ath11k(O) xt_DSCP wireguard pptp pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat_ftp nf_nat nf_flow_table_inet nf_flow_table nf_conntrack_netlink nf_conntrack_ftp nf_conntrack_bridge nf_conntrack mmc_spi mac80211(O) libchacha20poly1305 ipt_REJECT chacha_neon cfg80211(O) xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_CLASSIFY tmp103 spidev spi_gpio spi_bitbang slhc sch_cake rfcomm qrtr_smd qrtr_mhi qrtr qmi_helpers(O) poly1305_neon of_mmc_spi nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_quota nft_queue nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_compat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4
[  103.954460]  nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_dup_netdev nf_defrag_ipv4 mhi libcurve25519_generic libchacha iptable_mangle iptable_filter ipt_ECN ip6_tables ip_tables hidp hci_uart crc7 crc_itu_t compat(O) btusb btrtl btmtk btintel bnep bluetooth fuse ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs hid evdev input_core gpio_fan qca_nss_bridge_mgr(O) qca_nss_vlan(O) xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink msdos ip6_gre ip_gre gre ifb nat46(O) nf_defrag_ipv6 sit sctp qca_nss_drv(O) l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun hfsplus cdrom cifs oid_registry nls_ucs2_utils
[  104.044039]  cifs_md4 cifs_arc4 asn1_decoder dns_resolver netfs nls_utf8 nls_iso8859_15 nls_iso8859_1 nls_cp850 nls_cp437 nls_cp1250 rfkill wp512 twofish_generic twofish_common tea serpent_generic khazad cast6_generic cast5_generic cast_common camellia_generic blowfish_generic blowfish_common anubis ecdh_generic ecc xxhash_generic crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha512_arm64 sha2_ce sha256_arm64 sha1_ce seqiv sha3_generic jitterentropy_rng drbg michael_mic md5 kpp hmac geniv des_generic cmac blake2b_generic uas usb_storage leds_lp5569 leds_lp55xx_common leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ahci libahci libata qca_nss_dp(O) qca_ssdk(O) ramoops reed_solomon pstore gpio_button_hotplug(O) vfat fat f2fs ext4 mbcache jbd2 exfat btrfs xor xor_neon raid6_pq libcrc32c dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax aquantia hwmon crc_ccitt tpm rng_core crc32c_generic crc32_generic encrypted_keys trusted
[  104.219034] CPU: 1 PID: 2890 Comm: hostapd Tainted: G           O       6.6.63 #0
[  104.241269] Hardware name: Zyxel NBG7815 (DT)
[  104.248641] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  104.252985] pc : drv_remove_interface+0x70/0xc4 [mac80211]
[  104.259754] lr : drv_remove_interface+0x70/0xc4 [mac80211]
[  104.265309] sp : ffffffc08590b6c0
[  104.270772] x29: ffffffc08590b6c0 x28: 0000000000000065 x27: 0000000000000000
[  104.274166] x26: 0000000000000000 x25: 0000000000000001 x24: 0000000000000000
[  104.281284] x23: ffffffc08590b750 x22: 0000000000000001 x21: ffffffc08590b750
[  104.288403] x20: ffffff80139d08c0 x19: ffffff801a288900 x18: ffffffc080c2c328
[  104.295519] x17: 0000000000000002 x16: 0000000000000000 x15: 00000000000001fd
[  104.302637] x14: 00000000000001fd x13: 00000000ffffffea x12: ffffffc080c842d0
[  104.309756] x11: ffffffc080c2c328 x10: ffffffc080c84328 x9 : 0000000000000001
[  104.316874] x8 : 0000000000000001 x7 : 0000000000017fe8 x6 : c0000000ffffefff
[  104.323991] x5 : ffffff803fda5850 x4 : 0000000000000000 x3 : 0000000000000027
[  104.331110] x2 : 0000000000000027 x1 : ffffff80031ed580 x0 : 0000000100000000
[  104.338230] Call trace:
[  104.345337]  drv_remove_interface+0x70/0xc4 [mac80211]
[... cut ...]
[14536.649160]  kthread+0x10c/0x110
[14536.652806]  ret_from_fork+0x10/0x20
[14536.656192] ---[ end trace 0000000000000000 ]---
[14536.660467] ------------[ cut here ]------------
```

3.) Then I've faced issues with clients never getting an IP. In the past I've thought it is an DNSMASQ issue. Because they were introducing multiple instances (and they still have some issues with this setup) and I gave up with this kind of setup.

If there is sth. to test let me know. I like this approach very much.

Kind regards.

BrainSlayer, 2025-03-09:
@qosmio. can you please debug the following stes. check in drv_sta_state if old_state and new_state maches the condition for get_bss_data. because if get_bss_data isnt called there, check_sdata_in_driver will fail and triggers the bug. so it might be a state problem here or the whole condition with !SUPPORTS_NSS_OFFLOAD is wrong

qosmio, 2025-03-09:
@BrainSlayer added a bunch of debugging to driver-ops

```node
Sun Mar  9 14:36:38 2025 daemon.info hostapd: wlan0: STA da:e5:a8:ed:4f:27 IEEE 802.11: authenticated
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.159831] wlan0: Allocated STA da:e5:a8:ed:4f:27
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.159868] wlan0: state NOTEXIST -> NONE
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.159876] wlan0: sdata->vif.type: AP
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.159886] wlan0: sdata->vif.driver_flags: UAPSD
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.159892] wlan0: sdata->vif.offload_flags: ENCAP DECAP
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.159899] wlan0: sdata->flags: IN_DRIVER
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.160253] ath11k c000000.wifi: mac ath11k_peer_add_ast peer da:e5:a8:ed:4f:27 ast_entry da:e5:a8:ed:4f:27, ast_type 1
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.160275] ath11k c000000.wifi: mac ath11k_peer_map_ast peer da:e5:a8:ed:4f:27 ast_entry da:e5:a8:ed:4f:27
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.160294] ath11k c000000.wifi: mac Added peer: da:e5:a8:ed:4f:27 for VDEV: 1
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.160519] wlan0: Inserted STA da:e5:a8:ed:4f:27
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164462] wlan0: moving STA da:e5:a8:ed:4f:27 to state (AUTH)
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164475] wlan0: state NONE -> AUTH
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164483] wlan0: sdata->vif.type: AP
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164490] wlan0: sdata->vif.driver_flags: UAPSD
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164496] wlan0: sdata->vif.offload_flags: ENCAP DECAP
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164503] wlan0: sdata->flags: IN_DRIVER
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164513] wlan0: moving STA da:e5:a8:ed:4f:27 to state 3 (ASSOC)
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164520] wlan0: state AUTH -> ASSOC
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164527] wlan0: sdata->vif.type: AP
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164533] wlan0: sdata->vif.driver_flags: UAPSD
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164539] wlan0: sdata->vif.offload_flags: ENCAP DECAP
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164545] wlan0: sdata->flags: IN_DRIVER (interface was added to driver)
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.164558] ath11k c000000.wifi: mac peer da:e5:a8:ed:4f:27 phymode 11ax-he80
Sun Mar  9 14:36:38 2025 daemon.info hostapd: wlan0: STA da:e5:a8:ed:4f:27 IEEE 802.11: associated (aid 1)
Sun Mar  9 14:36:38 2025 daemon.notice hostapd: nl80211: wlan0[137]: set_sta_vlan(da:e5:a8:ed:4f:27, ifname=wlan0[137], vlan_id=0)
Sun Mar  9 14:36:38 2025 daemon.notice hostapd: Assigned VLAN ID 20 from wpa_psk_file to da:e5:a8:ed:4f:27
Sun Mar  9 14:36:38 2025 daemon.notice hostapd: nl80211: wlan0[137]: set_sta_vlan(da:e5:a8:ed:4f:27, ifname=wlan0-vlan20[138], vlan_id=20)
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.275360] ath11k c000000.wifi: <?> vif type 3, encap type 2
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.281467] ath11k c000000.wifi: <?> vif type 3, encap type 2
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.325416] wlan0-vlan20: moving STA da:e5:a8:ed:4f:27 to state 4 (AUTHORIZED)
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.325440] wlan0-vlan20: state ASSOC -> AUTHORIZED
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.325448] wlan0-vlan20: sdata->vif.type: AP_VLAN
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.325455] wlan0-vlan20: sdata->vif.driver_flags: NONE
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.325462] wlan0-vlan20: sdata->vif.offload_flags: NONE
Sun Mar  9 14:36:38 2025 kern.debug kernel: [39122.325469] wlan0-vlan20: sdata->flags: ALLMULTI (interface wants all multicast packets)
Sun Mar  9 14:36:38 2025 kern.warn kernel: [39122.325476] ------------[ cut here ]------------
Sun Mar  9 14:36:38 2025 kern.warn kernel: [39122.325508] wlan0-vlan20: Failed check-sdata-in-driver check, flags: 0x1
Sun Mar  9 14:36:38 2025 kern.warn kernel: [39122.329211] WARNING: CPU: 2 PID: 1461 at ../mac80211-regular/backports-6.12.6/net/mac80211/driver-ops.c:264 drv_sta_state+0x9d4/0xa04 [mac80211]
```

I tried to also follow the path in dev creation but can't pinpoint why NSS can create ext vdev for WDS just fine but freaks out about VLAN.

BrainSlayer, 2025-03-10:
your log does not contain debug info of old_state and new_state in drv_sta_state

BrainSlayer, 2025-03-10:
please also check if supports_nss_offload is set in drv_sta_state. your log says that sdata->flags does not contain the in_driver flag.
but the check fails. the check for nss_offload must fail or the state variable check fails for oldstate and newstate, so a wrong sdata is used. sdata = get_bss_sdata should not be called here, but it is as it looks

qosmio, 2025-03-11:
@BrainSlayer 
> your log does not contain debug info of old_state and new_state in drv_sta_state or maybe does it? (ASSOC -> AUTHORIZED)

I used the "->" notation to show `from` -> `to` state changes.

Excerpt of my debug statements.

```c

	might_sleep();
	lockdep_assert_wiphy(local->hw.wiphy);

	sta_dbg(sdata, "state: %s -> %s\n",
		ieee80211_sta_state_str(old_state), ieee80211_sta_state_str(new_state));
	sta_dbg(sdata, "sdata->vif.type: %s\n",
		ieee80211_vif_type_str(sdata->vif.type));
	sta_dbg(sdata, "sdata->vif.driver_flags: %s\n",
		ieee80211_vif_flags_str(sdata->vif.driver_flags, driver_flags_buf, sizeof(driver_flags_buf)));
	sta_dbg(sdata, "sdata->vif.offload_flags: %s\n",
		ieee80211_offload_flags_str(sdata->vif.offload_flags, offload_flags_buf, sizeof(offload_flags_buf)));
	sta_dbg(sdata, "sdata->flags: %s\n",
		ieee80211_sdata_flags_str(sdata->flags, sdata_flags_buf, sizeof(sdata_flags_buf)));

	if (!ieee80211_hw_check(&local->hw, SUPPORTS_NSS_OFFLOAD) ||
	    !(old_state == IEEE80211_STA_ASSOC &&
	      new_state == IEEE80211_STA_AUTHORIZED)) {
		if (!ieee80211_hw_check(&local->hw, SUPPORTS_NSS_OFFLOAD))
			sdata_info(sdata,
				"get_bss_sdata called: !(SUPPORTS_NSS_OFFLOAD)\n");
		else
			sdata_info(sdata,
				"get_bss_sdata called: !(ASSOC -> AUTHORIZED) (old_state=%d, new_state=%d)\n",
				old_state, new_state);

		sdata = get_bss_sdata(sdata);
		sta_dbg(sdata, "get_bss_sdata sdata->vif.driver_flags: %s\n",
				ieee80211_vif_flags_str(sdata->vif.driver_flags, driver_flags_buf, sizeof(driver_flags_buf)));
		sta_dbg(sdata, "get_bss_sdata sdata->vif.offload_flags: %s\n",
				ieee80211_offload_flags_str(sdata->vif.offload_flags, offload_flags_buf, sizeof(offload_flags_buf)));
		sta_dbg(sdata, "get_bss_sdata sdata->flags: %s\n",
				ieee80211_sdata_flags_str(sdata->flags, sdata_flags_buf, sizeof(sdata_flags_buf)));
	}

	if (!check_sdata_in_driver(sdata))
		return -EIO;

```

```c
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.464731] phy0-ap1: Allocated STA 80:69:1a:22:97:78
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.464769] phy0-ap1: state: NOTEXIST -> NONE
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.464777] phy0-ap1: sdata->vif.type: AP
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.464785] phy0-ap1: sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.464792] phy0-ap1: sdata->vif.offload_flags: ENCAP DECAP
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.464800] phy0-ap1: sdata->flags: ALLMULTI NO_BRIDGE SDATA_IN_DRIVER
Tue Mar 11 02:13:32 2025 kern.info kernel: [14858.464807] phy0-ap1: get_bss_sdata called: !(ASSOC -> AUTHORIZED) (old_state=0, new_state=1)
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.464816] phy0-ap1: get_bss_sdata sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.464823] phy0-ap1: get_bss_sdata sdata->vif.offload_flags: ENCAP DECAP
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.464830] phy0-ap1: get_bss_sdata sdata->flags: ALLMULTI NO_BRIDGE SDATA_IN_DRIVER
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.465525] phy0-ap1: Inserted STA 80:69:1a:22:97:78
Tue Mar 11 02:13:32 2025 daemon.info hostapd: phy0-ap1: STA 80:69:1a:22:97:78 IEEE 802.11: authenticated
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482705] phy0-ap1: moving STA 80:69:1a:22:97:78 to state 2
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482735] phy0-ap1: state: NONE -> AUTH
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482743] phy0-ap1: sdata->vif.type: AP
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482751] phy0-ap1: sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482758] phy0-ap1: sdata->vif.offload_flags: ENCAP DECAP
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482766] phy0-ap1: sdata->flags: ALLMULTI NO_BRIDGE SDATA_IN_DRIVER
Tue Mar 11 02:13:32 2025 kern.info kernel: [14858.482772] phy0-ap1: get_bss_sdata called: !(ASSOC -> AUTHORIZED) (old_state=1, new_state=2)
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482780] phy0-ap1: get_bss_sdata sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482787] phy0-ap1: get_bss_sdata sdata->vif.offload_flags: ENCAP DECAP
[... 2 lines cut ...]
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482812] phy0-ap1: state: AUTH -> ASSOC
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482818] phy0-ap1: sdata->vif.type: AP
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482824] phy0-ap1: sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482831] phy0-ap1: sdata->vif.offload_flags: ENCAP DECAP
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482837] phy0-ap1: sdata->flags: ALLMULTI NO_BRIDGE SDATA_IN_DRIVER
Tue Mar 11 02:13:32 2025 kern.info kernel: [14858.482844] phy0-ap1: get_bss_sdata called: !(ASSOC -> AUTHORIZED) (old_state=2, new_state=3)
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482850] phy0-ap1: get_bss_sdata sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482857] phy0-ap1: get_bss_sdata sdata->vif.offload_flags: ENCAP DECAP
Tue Mar 11 02:13:32 2025 kern.debug kernel: [14858.482863] phy0-ap1: get_bss_sdata sdata->flags: ALLMULTI NO_BRIDGE SDATA_IN_DRIVER
Tue Mar 11 02:13:32 2025 daemon.info hostapd: phy0-ap1: STA 80:69:1a:22:97:78 IEEE 802.11: associated (aid 1)
[... cut ...]
Tue Mar 11 02:13:32 2025 kern.warn kernel: [14858.586006] phy0-ap1-20: Failed check-sdata-in-driver check, flags: 0x1
Tue Mar 11 02:13:32 2025 kern.warn kernel: [14858.589845] WARNING: CPU: 0 PID: 28815 at ../mac80211-regular/backports-6.12.6/net/mac80211/driver-ops.c:288 drv_sta_state+0x33c/0x578 [mac80211]
```

So `get_bss_sdata` is called for state changes `NOTEXIST -> NONE`, `NONE -> AUTH`, `AUTH -> ASSOC`. I tested removing the `!(old_state == IEEE80211_STA_ASSOC && new_state == IEEE80211_STA_AUTHORIZED))` check fully so it always skipped if NSS was enabled, but no go that route either.

BrainSlayer, 2025-03-11:
but from your log get_bss_sdata is called and thats the problem here.  oldstate = ASSOC, newstate = AUTHORIZED in your log. 
and nss is enabled. 

but in this condition get_bss_sdata should never be called. but it is called. otherwise since sdata->flags is ALLMULTI check_sdata_in_driver would not fail

qosmio, 2025-03-11:
Sorry, I was also saying that I tested removing the check completely, so it only checks if NSS offloading is disabled via `(!ieee80211_hw_check(&local->hw, SUPPORTS_NSS_OFFLOAD)):

```c
	if (!ieee80211_hw_check(&local->hw, SUPPORTS_NSS_OFFLOAD)) {
		pr_debug("%s: get_bss_sdata called: !(SUPPORTS_NSS_OFFLOAD)\n",
				sdata->name);

		sdata = get_bss_sdata(sdata);
	}

	if (!check_sdata_in_driver(sdata))
		return -EIO;
```

But results in the same thing.
```c
Tue Mar 11 13:29:56 2025 daemon.info hostapd: phy0-ap0: STA da:e5:a8:ed:4f:27 IEEE 802.11: authenticated
Tue Mar 11 13:29:56 2025 daemon.info hostapd: phy0-ap0: STA da:e5:a8:ed:4f:27 IEEE 802.11: associated (aid 1)
Tue Mar 11 13:29:56 2025 daemon.notice hostapd: nl80211: phy0-ap0[19]: set_sta_vlan(da:e5:a8:ed:4f:27, ifname=phy0-ap0[19], vlan_id=0)
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.643401] phy0-ap0: state: NOTEXIST -> NONE
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.643427] phy0-ap0: sdata->vif.type: AP
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.643436] phy0-ap0: sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.643443] phy0-ap0: sdata->vif.offload_flags: ENCAP DECAP
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.643450] phy0-ap0: sdata->flags: SDATA_IN_DRIVER
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647877] phy0-ap0: state: NONE -> AUTH
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647902] phy0-ap0: sdata->vif.type: AP
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647909] phy0-ap0: sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647917] phy0-ap0: sdata->vif.offload_flags: ENCAP DECAP
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647924] phy0-ap0: sdata->flags: SDATA_IN_DRIVER
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647934] phy0-ap0: state: AUTH -> ASSOC
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647940] phy0-ap0: sdata->vif.type: AP
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647946] phy0-ap0: sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647953] phy0-ap0: sdata->vif.offload_flags: ENCAP DECAP
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.647959] phy0-ap0: sdata->flags: SDATA_IN_DRIVER
Tue Mar 11 13:29:56 2025 daemon.notice hostapd: Assigned VLAN ID 20 from wpa_psk_file to da:e5:a8:ed:4f:27
Tue Mar 11 13:29:56 2025 daemon.notice hostapd: nl80211: phy0-ap0[19]: set_sta_vlan(da:e5:a8:ed:4f:27, ifname=phy0-ap0-20[20], vlan_id=20)
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.793519] phy0-ap0-20: state: ASSOC -> AUTHORIZED
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.793542] phy0-ap0-20: sdata->vif.type: AP_VLAN
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.793551] phy0-ap0-20: sdata->vif.driver_flags: NONE
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.793558] phy0-ap0-20: sdata->vif.offload_flags: NONE
Tue Mar 11 13:29:56 2025 kern.debug kernel: [  607.793565] phy0-ap0-20: sdata->flags: ALLMULTI
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.793573] ------------[ cut here ]------------
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.793605] phy0-ap0-20: Failed check-sdata-in-driver check, flags: 0x1
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.797363] WARNING: CPU: 0 PID: 7082 at ../mac80211-regular/backports-6.12.6/net/mac80211/driver-ops.c:294 drv_sta_state+0x2cc/0x558 [mac80211]
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.926064] CPU: 0 PID: 7082 Comm: hostapd Tainted: G           O       6.6.82 #0
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.948300] Hardware name: Linksys MX4300 (DT)
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.955846] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.960190] pc : drv_sta_state+0x2cc/0x558 [mac80211]
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.967044] lr : drv_sta_state+0x2cc/0x558 [mac80211]
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.972252] sp : ffffffc0840834a0
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.977284] x29: ffffffc084083630 x28: 0000000000000065 x27: 0000000000000000
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.980590] x26: ffffff801345c900 x25: ffffffc07950a7a0 x24: ffffff8013694000
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.987707] x23: 0000000000000003 x22: 0000000000000004 x21: ffffff801d9d08c0
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  607.994825] x20: ffffffc07950a700 x19: ffffff801345c900 x18: ffffffc080c9a640
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  608.001944] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000846
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  608.009062] x14: 0000000000000846 x13: 00000000ffffffea x12: ffffffc080cf25e8
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  608.016180] x11: ffffffc080c9a640 x10: ffffffc080cf2640 x9 : 0000000000000001
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  608.023297] x8 : 0000000000000001 x7 : 0000000000017fe8 x6 : 0000000000000001
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  608.030415] x5 : ffffff807fb93850 x4 : 0000000000000000 x3 : 0000000000000027
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  608.037533] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 000000000000003b
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  608.044653] Call trace:
Tue Mar 11 13:29:56 2025 kern.warn kernel: [  608.051761]  drv_sta_state+0x2cc/0x558 [mac80211]
[... cut ...]
Tue Mar 11 13:39:36 2025 kern.debug kernel: [ 1188.138893] phy0-ap0: sdata->vif.driver_flags: SUPPORTS_UAPSD
Tue Mar 11 13:39:36 2025 kern.debug kernel: [ 1188.138900] phy0-ap0: sdata->vif.offload_flags: ENCAP DECAP
Tue Mar 11 13:39:36 2025 kern.debug kernel: [ 1188.138908] phy0-ap0: sdata->flags: SDATA_IN_DRIVER
Tue Mar 11 13:39:36 2025 kern.warn kernel: [ 1188.138931] ath11k c000000.wifi: Failed to add peer: da:e5:a8:ed:4f:27 for VDEV: 0
Tue Mar 11 13:39:36 2025 kern.warn kernel: [ 1188.138983] ath11k c000000.wifi: Failed to add station: da:e5:a8:ed:4f:27 for VDEV: 0
```

BrainSlayer, 2025-03-12:
removing the check is wrong too. this function is called several times. on first attempt it must use get_bss_sdata(sdata);. it just needs to be ignored when turning to authorized. so find out why this condition doesnt work. thats the one and only question 

its using the wrong sdata pointer. removing the check results in still using the wrong sdata pointer, just at another call

and still from your log. check_sdata_in_driver shows that the in_driver flag is set so it warns. your log shows the in_Driver flag is "not set" but just ALLMULTI. find out why this happens. something is very wrong here. a wrong pointer is used.

qosmio, 2025-03-14:
> it just needs to be ignored when turning to authorized. so find out why this condition doesnt work. thats the one and only question

... that's the question I'm struggling to trace. I'm not sure where the pointer for it is being set between the calls from mac80211 and NSS. It seems that SDATA_IN_DRIVER is always null when iftype is AP_VLAN (even non-NSS), but non-offload path uses `sdata = container_of(sdata->bss, struct ieee80211_sub_if_data, u.ap)` to get parent interface of the AP, but I don't see anywhere in the patches that indicates what the equivalent behavior should be with NSS.

qosmio, 2026-03-01:
@tardeaux, just tested it and well whadya know... it works. Well done! 👍 

Looking it over, I'm surprised I missed something so obvious. I was experimenting by forcing the connection in `ieee80211_do_open` regardless of `sdata->bss->active`.
```diff
--- a/net/mac80211/driver-ops.c
+++ b/net/mac80211/driver-ops.c
@@ -510,7 +510,10 @@ int drv_set_key(struct ieee80211_local *local,
 	might_sleep();
 	lockdep_assert_wiphy(local->hw.wiphy);
 
-	sdata = get_bss_sdata(sdata);
+	if (!ieee80211_hw_check(&local->hw, SUPPORTS_NSS_OFFLOAD) ||
+	    sdata->vif.type != NL80211_IFTYPE_AP_VLAN)
+		sdata = get_bss_sdata(sdata);
+
 	if (!check_sdata_in_driver(sdata))
 		return -EIO;
--- a/net/mac80211/iface.c
+++ b/net/mac80211/iface.c
@@ -1439,6 +1439,13 @@ int ieee80211_do_open(struct wireless_dev *wdev, bool coming_up)
 			ieee80211_set_vif_encap_ops(sdata);
 		} else {
 			netif_carrier_off(dev);
+
+			if (ieee80211_hw_check(&local->hw, SUPPORTS_NSS_OFFLOAD)) {
+				ieee80211_set_sdata_offload_flags(sdata);
+				res = drv_add_interface(local, sdata);
+				if (res)
+					goto err_del_interface;
+			}
```

Curious about this though. Is this check needed during `drv_vif_add_debugfs`?
```diff
--- a/net/mac80211/driver-ops.h
+++ b/net/mac80211/driver-ops.h
@@ -521,7 +521,9 @@ static inline void drv_vif_add_debugfs(struct ieee80211_local *local,
 	    WARN_ON(!sdata->vif.debugfs_dir))
 		return;

-	sdata = get_bss_sdata(sdata);
+	if (!(sdata->vif.type == NL80211_IFTYPE_AP_VLAN &&
+	      ieee80211_hw_check(&local->hw, SUPPORTS_NSS_OFFLOAD)))
+		sdata = get_bss_sdata(sdata);
 	if (!check_sdata_in_driver(sdata))
 		return;
```