qosmio/openwrt-ipq #35: nss-24.10 fails to compile due to ath11k patch failure

motolav, 2025-04-21:
### Describe the bug

The nss-24.10 branch fails to compile after the update for 24.10.1 due to a ath11k patch
```
Applying /openwrt-ipq-24.10-nss/package/kernel/mac80211/patches/nss/ath11k/373-ath11k-Add-retry-mechanism-for-update_rx_qu.patch using plaintext: 
patching file drivers/net/wireless/ath/ath11k/core.h
patching file drivers/net/wireless/ath/ath11k/debugfs.c
patching file drivers/net/wireless/ath/ath11k/dp.c
Hunk #1 succeeded at 1102 (offset -13 lines).
patching file drivers/net/wireless/ath/ath11k/dp.h
Hunk #1 succeeded at 23 with fuzz 1 (offset -1 lines).
Hunk #2 succeeded at 54 (offset 2 lines).
Hunk #3 succeeded at 306 (offset 2 lines).
patching file drivers/net/wireless/ath/ath11k/dp_rx.c
Hunk #5 FAILED at 805.
Hunk #8 FAILED at 900.
2 out of 8 hunks FAILED -- saving rejects to file drivers/net/wireless/ath/ath11k/dp_rx.c.rej
```

### OpenWrt version

nss-24.10

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ipq807x

### Device

Dynalink

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

alth0, 2025-04-21:
I am experiencing this as well, happens exactly at this specific ath11k patch with identical output.

qosmio, 2025-04-25:
The issue was resolved in commit a4b294fdb7e0622b74fba8e506b20609a032a85d `ath11k_nss: Fix patch against "Fix DMA buffer allocation to resolve SWIOTLB issues"`

Please try against latest `24.10-nss` commit (f667beb4832)

alth0, 2025-04-25:
The build just finished smoothly without any errors after the latest commit. Thank you.

closed 2025-04-25 by qosmio