qosmio/openwrt-ipq #36: ipq6018-nss.dtsi error

try496, 2025-04-26:
### Describe the bug

Use this nss to work and take effect https://github.com/JiaY-shi/openwrt/commit/35f2bbb0e819fd81a8ea90565b65aaa8532cb390

### OpenWrt version

r29403-14e9c1f936

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq60xx

### Device

Redmi AX5

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

qosmio, 2025-05-01:
Can you verify if this is still an issue? Added dummy NSS regulator

d429490aba5342a5282af158c49d36efa9c65d1f

try496, 2025-05-01:
> 您能否验证这是否仍然是一个问题？添加了虚拟 NSS 稳压器
> 
> [编号 d429490](https://github.com/qosmio/openwrt-ipq/commit/d429490aba5342a5282af158c49d36efa9c65d1f)

yes

VIKINGYFY, 2025-05-02:
> Can you verify if this is still an issue? Added dummy NSS regulator
> 
> [d429490](https://github.com/qosmio/openwrt-ipq/commit/d429490aba5342a5282af158c49d36efa9c65d1f)

The cause of this problem is the nodes under the IPQ60XX soc
`#address-cells = <2>;`
`#size-cells = <2>;`
So it is necessary to
`reg = <0x01868010 0x1000>, <0x40000000 0x1000>;`
`reg = <0x39000000 0x1000>, <0x0b111000 0x1000>;`
Modify to
`reg = <0x0 0x01868010 0x0 0x1000>, <0x0 0x40000000 0x0 0x1000>;`
`reg = <0x0 0x39000000 0x0 0x1000>, <0x0 0x0b111000 0x0 0x1000>;`

qosmio, 2025-05-30:
closing as it no longer appears when compiling

closed 2025-05-30 by qosmio