qosmio/openwrt-ipq #45: ipq5018 nss support

Art-Chen, 2025-05-10:
### Describe the bug

ipq5018 can not get nss working normal.

# Build Break
build break when applying patch for `qca-nss-drv` when using FW 11.4.
```
Applying /home/chenxy/openwrt-6.6-ipq5018/feeds/nss_packages/qca-nss-drv/patches-11.4/0069-qca-nss-drv-Return-MTU-as-is-from-IPQ50XX-HAL.patch using plaintext:
make[3]: Entering directory '/home/chenxy/openwrt-6.6-ipq5018/feeds/luci/applications/luci-app-sqm'
patching file nss_data_plane/hal/nss_ipq50xx.c
Hunk #1 FAILED at 1.
Hunk #2 FAILED at 161.
2 out of 2 hunks FAILED -- saving rejects to file nss_data_plane/hal/nss_ipq50xx.c.rej
Patch failed!  Please fix /home/chenxy/openwrt-6.6-ipq5018/feeds/nss_packages/qca-nss-drv/patches-11.4/0069-qca-nss-drv-Return-MTU-as-is-from-IPQ50XX-HAL.patch!
make[3]: *** [Makefile:370: /home/chenxy/openwrt-6.6-ipq5018/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/qca-nss-drv-11.4.0.5.2021.09.13~53e5863/.prepared_7e12d1ac8f9e003dc4c52b8da95c75f4_6664517399ebbbc92a37c5bb081b5c53] Error 1
```

after switch to 12.2/12.5, the build is successful.

# Runtime Issues
NX54 has a 512M ram. when using ath11k 512M memory layout, we got panic when firmware ready.
see the dmesg
[dmesg_512M.txt](https://github.com/user-attachments/files/20139507/dmesg_512M.txt)

after switch to use 256M memory layout, we can successfully booted into system.
but all the `lan` can not transfer data(phy detect working, but tx/rx is zero). `wan` works well and can connected to the internet via DHCP, issues disappear after disable nss.

no error on the dmesg
[dmesg_256M.txt](https://github.com/user-attachments/files/20139458/dmesg_256M.txt)

is any idea about these issues?

### OpenWrt version

r29664+8-a0ea0a8e6d

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq50xx

### Device

H3C NX54

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Art-Chen, 2025-05-10:
try communicate router via `wan`, after testing iperf3, the speed is zero, and `wan led` keeps blinking even if stop the test(tx/rx activity).

qosmio, 2025-05-31:
```
> Applying /home/chenxy/openwrt-6.6-ipq5018/feeds/nss_packages/qca-nss-drv/patches-11.4/0069-qca-nss-drv-Return-MTU-as-is-from-IPQ50XX-HAL.patch using plaintext:
```

This patch `qca-nss-drv/patches-11.4/0069-qca-nss-drv-Return-MTU-as-is-from-IPQ50XX-HAL.patch` isn't from my repository. I also don't know the contents of your patch or the NSS repo you're using, or what previous patches you've applied...

I'm assuming based on your patch name `Return-MTU-as-is-from-IPQ50XX` it's supposed to be similar to:
[0030-backport-12.5-ipq50xx-fixes.patch
](https://raw.githubusercontent.com/qosmio/nss-packages/refs/heads/NSS-12.5-K6.x/qca-nss-drv/patches-11.4/0030-backport-12.5-ipq50xx-fixes.patch)

```diff
@@ -161,25 +140,7 @@ void nss_data_plane_hal_stats_sync(struc
 uint16_t nss_data_plane_hal_get_mtu_sz(uint16_t mtu)
 {
 	/*
-	 * GMACs support 3 Modes
-	 * Normal Mode Payloads upto 1522 Bytes ( 1500 + 14 + 4(Vlan) + 4(CRC))
-	 * Mini Jumbo Mode Payloads upto 2000 Bytes (1978 + 14 + 4(Vlan) + 4 (CRC))
-	 * Full Jumbo Mode payloads upto 9022 Bytes (9000 + 14 + 4(Vlan) + 4 (CRC))
-	 */
-
-	/*
-	 * The configured MTU value on a GMAC interface should be one of these
-	 * cases. Finding the Needed MTU size that is required for GMAC to
-	 * successfully receive the frame.
+	 * Return MTU value as is.
 	 */
-	if (mtu <= NSS_DP_GMAC_NORMAL_FRAME_MTU) {
-	    return NSS_DP_GMAC_NORMAL_FRAME_MTU;
-	}
-	if (mtu <= NSS_DP_GMAC_MINI_JUMBO_FRAME_MTU) {
-	    return NSS_DP_GMAC_MINI_JUMBO_FRAME_MTU;
-	}
-	if (mtu <= NSS_DP_GMAC_FULL_JUMBO_FRAME_MTU) {
-		return NSS_DP_GMAC_FULL_JUMBO_FRAME_MTU;
-	}
-	return 0;
+	return mtu;
 }
```

Also, I'm not sure I can be of much help. IPQ5018 has been very difficult to get working with NSS. I haven't seen anyone with a working build. It's something very strange with the hardware.

closed 2025-07-15 by qosmio

Art-Chen, 2025-07-26:
@qosmio is any option to get more log about nss? i can not found any more error there, so I'm not sure we're missing anything. or Can you tell me how you determine that ipq807x/ipq70xx needs certain patches?