qosmio/openwrt-ipq #46: Boot Loop on Arcadyan AW1000 (IPQ807x) with qosmio/openwrt-ipq - 24.10-nss Build

intannajwa, 2025-05-21:
Hi, I successfully built the firmware using your qosmio/openwrt-ipq - 24.10-nss for the Arcadyan AW1000 (IPQ807x). However, after flashing it to my modem, the device enters a boot loop and keeps restarting. Interestingly, when I flash the official ImmortalWRT build, it works fine without any issues. Could you please help me figure out what might be going wrong with the custom build or suggest how to fix it? Thank you!

intannajwa, 2025-05-21:
I’ve identified the issue: the system enters a boot loop when using quectel-cm with the Quectel protocol. It works normally when using other protocols.

Razifadm, 2025-05-24:
you need to use wwan package from immortalwrt.

Razifadm, 2025-05-26:
hi, care to share how you fixed the cpu0 spike? much appreciated of thanksss

qosmio, 2025-05-31:
Folks,

I've [removed all `wwan` packages from my primary `nss-packages`](https://github.com/qosmio/nss-packages/commit/d845680ef287bdef12553224023110c689f66c54) feed. It should have been kept separate but I added it a while back for testing purposes.

Since I personally don't have this device, I can't be of much help.

I'll leave this issue up for a few days if @intannajwa wants to respond with their solution.

cjayacopra, 2025-05-31:
Hi @qosmio,

I'm currently working on a build for my AW1000 and encountered a problem today: I'm unable to install quectel-cm. I would greatly appreciate it if you could make it available again.

Regarding the solution, I suggest configuring a wwan interface to act as a DHCP client for the quectel-cm interface. This method leverages the extensive work and improvements already integrated into the wwan contributions within your nss-packages by Hikari and other developers, which is a more robust approach than using wwan from immortalwrt."

intannajwa, 2025-06-01:
> > > hi, care to share how you fixed the cpu0 spike? much appreciated of thanksss
> > 
> > 
> > I'll back up the complete settings and upload them later. I'm outside right now, maybe tonight.
> 
> hi, thank for the reply. waiting for your golden approach to save my marathon cpu0. hahaha. looking forward for your repliesssss thanksss

Sorry for the delay — I forgot.
Here's my simple setup. It's not balancing the CPU usage to 100%, but I hope you can improve it: https://github.com/intannajwa/bk-ipq

intannajwa, 2025-06-01:
> Hi [@qosmio](https://github.com/qosmio),
> 
> I'm currently working on a build for my AW1000 and encountered a problem today: I'm unable to install quectel-cm. I would greatly appreciate it if you could make it available again.
> 
> Regarding the solution, I suggest configuring a wwan interface to act as a DHCP client for the quectel-cm interface. This method leverages the extensive work and improvements already integrated into the wwan contributions within your nss-packages by Hikari and other developers, which is a more robust approach than using wwan from immortalwrt. Enabling all DRV in the config should fix the bootloop as indicated in the OpenWrt forum for AW1000.

use https://github.com/immortalwrt/wwan-packages

intannajwa, 2025-06-01:
> Folks,
>
> I've [removed all `wwan` packages from my primary `nss-packages`](https://github.com/qosmio/nss-packages/commit/d845680ef287bdef12553224023110c689f66c54) feed. It should have been kept [...]

Thanks

intannajwa, 2025-06-01:
for auto set irq use ``#!/bin/sh

set_affinity() {
    IRQ=$1
    CPU_MASK=$2
    LABEL=$3
    echo "$CPU_MASK" > /proc/irq/$IRQ/smp_affinity
    echo "Set $LABEL IRQ $IRQ to CPU mask $CPU_MASK"
}

# CPU1
USB3_IRQ=42
set_affinity "$USB3_IRQ" 2 "USB3"
 
NSSQ0_IRQ=$(grep "nss_queue0" /proc/interrupts | head -n 1 | awk '{print $1}' | tr -d :)
NSSQ1_IRQ=$(grep "nss_queue1" /proc/interrupts | head -n 1 | awk '{print $1}' | tr -d :)

if [ -n "$NSSQ0_IRQ" ]; then
    set_affinity "$NSSQ0_IRQ" 4 "NSS Queue0"  # CPU2
else
    echo "NSS Queue0 IRQ not found"
fi

if [ -n "$NSSQ1_IRQ" ]; then
    set_affinity "$NSSQ1_IRQ" 8 "NSS Queue1"  # CPU3
else
    echo "NSS Queue1 IRQ not found"
fi
``

intannajwa, 2025-06-01:
Or Use round-robin distribution across CPUs for all NSS and USB-related IRQs 

```
#!/bin/sh

set_affinity() {
    IRQ=$1
    CPU_MASK=$2
    LABEL=$3

    if [ -e /proc/irq/"$IRQ"/smp_affinity ]; then
        echo "$CPU_MASK" > /proc/irq/"$IRQ"/smp_affinity
        echo "Set $LABEL IRQ $IRQ to CPU mask $CPU_MASK"
    else
        echo "IRQ $IRQ ($LABEL) not found or inaccessible"
    fi
}
 
cpu=0
get_mask() {
    case $1 in
        0) echo 1 ;;  # CPU0
        1) echo 2 ;;  # CPU1
        2) echo 4 ;;  # CPU2
        3) echo 8 ;;  # CPU3
    esac
}
 
grep -i "nss_queue" /proc/interrupts | while read -r line; do
    IRQ=$(echo "$line" | awk '{print $1}' | tr -d :)
    LABEL=$(echo "$line" | awk '{print $NF}')
    MASK=$(get_mask $cpu)
    set_affinity "$IRQ" "$MASK" "NSS $LABEL"
    cpu=$(( (cpu + 1) % 4 ))
done
 
grep -i "xhci" /proc/interrupts | while read -r line; do
    IRQ=$(echo "$line" | awk '{print $1}' | tr -d :)
    LABEL=$(echo "$line" | awk '{print $NF}')
    MASK=$(get_mask $cpu)
    set_affinity "$IRQ" "$MASK" "USB $LABEL"
    cpu=$(( (cpu + 1) % 4 ))
done
```

intannajwa, 2025-06-01:
or any batter one [@qosmio](https://github.com/qosmio),

qosmio, 2025-06-01:
> Or Use round-robin distribution across CPUs for all NSS and USB-related IRQs
>
> ```
> #!/bin/sh
>
> set_affinity() {
>     IRQ=$1
>     CPU_MASK=$2
>     LABEL=$3
>
>     if [ -e /proc/irq/"$IRQ"/smp_affinity ]; [...]

You don't need to run this if you're using the nss-drv package from my repository. I have a pretty [extensive init script](https://github.com/qosmio/nss-packages/blob/NSS-12.5-K6.x/qca-nss-drv/files/qca-nss-drv.init) that manages the affinities for everything.

The only thing that is likely missing is accounting for multiple XHCI. I only accounted for one USB IRQ.

What does your /proc/interrupts look like without running your script?

intannajwa, 2025-06-01:
> /proc/interrupts
hereis

```
cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3       
  9:          0          0          0          0 GIC-0  39 Level     arch_mem_timer
 13:       5561       4430       3835       4073 GIC-0  20 Level     arch_timer
 16:          0          0          0          0 GIC-0 239 Level     bam_dma
 17:          0          0          0          0 GIC-0 270 Level     bam_dma
 18:      89354          0          0          0 GIC-0 178 Level     bam_dma
 19:          4          0          0          0 GIC-0 354 Edge      smp2p
 20:          8          0          0          0 GIC-0 340 Level     msm_serial0
 22:          0          0          0          0 GIC-0 216 Level     4a9000.thermal-sensor
 23:          0          0          0          0 GIC-0  35 Edge      wdt_bark
 24:          0          0          0          0 GIC-0 357 Edge      q6v5 wdog
 25:          0          0          0          0  pmic_arb 51380237 Edge      pm-adc5
 26:          0          0          0          0 GIC-0  23 Level     arm-pmu
 27:         29          0          0          0 GIC-0  47 Edge      cpr3
 28:          0          0          0          0     smp2p   0 Edge      q6v5 fatal
 29:          2          0          0          0     smp2p   1 Edge      q6v5 ready
 31:          0          0          0          0     smp2p   2 Edge      q6v5 handover
 32:          0          0          0          0     smp2p   3 Edge      q6v5 stop
 33:          0          0          0          0   msmgpio   2 Edge      keys
 34:          0          0          0          0   msmgpio  67 Edge      keys
 35:          0          0          0          0 GIC-0 377 Level     edma_txcmpl
 36:          0          0          0          0 GIC-0 385 Level     edma_rxfill
 37:          0          0          0          0 GIC-0 393 Level     edma_rxdesc
 38:          0          0          0          0 GIC-0 376 Level     edma_misc
 39:          0          0          0          0 GIC-0 252 Level     qcom_dwc3 SS
 40:          0          0          0          0 GIC-0 257 Level     qcom_dwc3 SS
 41:          0          0          0          0 GIC-0 172 Level     xhci-hcd:usb1
 42:       1679          0          0          0 GIC-0 131 Level     xhci-hcd:usb3
 43:        252        620          0          0 GIC-0 409 Edge      nss_empty_buf_sos
 44:          4          0       1274          0 GIC-0 410 Edge      nss_empty_buf_queue
 45:          0          0          0          0 GIC-0 411 Edge      nss-tx-unblock
 46:      27586          0          0          0 GIC-0 412 Edge      nss_queue0
 47:          0        288          0          0 GIC-0 413 Edge      nss_queue1
 48:          0          0        129          0 GIC-0 414 Edge      nss_queue2
 49:          0          0          0        171 GIC-0 415 Edge      nss_queue3
 50:          0          0          0          0 GIC-0 416 Edge      nss_coredump_complete
 51:          0          0          0          0 GIC-0 417 Edge      nss_paged_empty_buf_sos
 52:          0          0          0          0 GIC-0 418 Edge      nss_profile_dma
 53:         61          0          0        171 GIC-0 422 Edge      nss_empty_buf_sos
 55:          0          0          0          0 GIC-0 423 Edge      nss_empty_buf_queue
 56:          0          0          0          0 GIC-0 424 Edge      nss-tx-unblock
 57:       4944          0          0      15529 GIC-0 425 Edge      nss_queue0
 58:          0          0          0          0 GIC-0 426 Edge      nss_queue1
 59:          0          0          0          0 GIC-0 427 Edge      nss_queue2
 60:          0          0          0          0 GIC-0 428 Edge      nss_queue3
 61:          0          0          0          0 GIC-0 429 Edge      nss_coredump_complete
 62:          0          0          0          0   msmgpio  62 Edge      keys
 64:          0          0          0          0 GIC-0 430 Edge      nss_paged_empty_buf_sos
 65:        128          0          0          0 GIC-0 353 Edge      glink-smem
 66:          5          0          0          0 GIC-0 348 Edge      ce0
 67:       1985          0          0          0 GIC-0 347 Edge      ce1
 68:       6131          0          0          0 GIC-0 346 Edge      ce2
 69:        375          0          0          0 GIC-0 343 Edge      ce3
 70:          0          0          0          0 GIC-0 443 Edge      ce5
 71:        301          0          0          0 GIC-0  72 Edge      ce7
 72:          0          0          0          0 GIC-0 334 Edge      ce9
 73:          0          0          0          0 GIC-0 333 Edge      ce10
 74:          0          0          0          0 GIC-0  69 Edge      ce11
 75:          0          0          0          0 GIC-0 209 Edge      rxdma2host-destination-ring-mac1
 76:          0          0          0          0 GIC-0 211 Edge      rxdma2host-destination-ring-mac3
 77:          0          0          0          0 GIC-0 210 Edge      rxdma2host-destination-ring-mac2
 78:          4          0          0          0 GIC-0 321 Edge      reo2host-status
 79:         18       4542          0          0 GIC-0 261 Edge      ppdu-end-interrupts-mac1
 80:          1          0          0          0 GIC-0 255 Edge      rxdma2host-monitor-status-ring-mac1
 81:         55          0       2746         10 GIC-0 263 Edge      ppdu-end-interrupts-mac3
[... cut ...]
IPI1:      5801      12438      11818       8390       Function call interrupts
IPI2:         0          0          0          0       CPU stop interrupts
IPI3:         0          0          0          0       CPU stop (for crash dump) interrupts
IPI4:         0          0          0          0       Timer broadcast interrupts
IPI5:       152        205        207        194       IRQ work interrupts
IPI6:         0          0          0          0       CPU wake-up interrupts
Err:          0
```

cjayacopra, 2025-06-01:
> You don't need to run this if you're using the nss-drv package from my repository. I have a pretty [extensive init [...]

That was actually the one causing bootlop. When nss-drv is not installed.

Razifadm, 2025-06-01:
> You don't need to run this if you're using the nss-drv package from my repository. I have a pretty [extensive init [...]

So from my pov, using your repo, doesn't need any irqbalance, or scrip tweak right? Like, "compile"-flash- voila nss. But cpu0 still spike. Looking forward for adding dhcp interface. As suggested above

qosmio, 2025-06-01:
I'll take a look later today.

Also , I saw you had irqbalance on but you disabled. It should'nt be ever be installed, and it's Apple about in the forums got log

intannajwa, 2025-06-01:
@qosmio  This qca-nss-drv/files/qca-nss-drv.init script works with my AW1000—no more CPU spikes!

```
#!/bin/sh /etc/rc.common
# vim: set syn=bash
# shellcheck disable=2155,3010,3019,3043,3057,3060

START=94
PROG="qca-nss-drv"

TOTAL_CPUS=$(grep -c ^processor /proc/cpuinfo)
[ "$TOTAL_CPUS" -lt 4 ] && CPU_LIST="0-$((TOTAL_CPUS-1))" || CPU_LIST="0-3"

log_msg() {
    logger -t "$PROG" "Pinning IRQ($4) $1 to CPU $2"
}

cpus_to_bitmask() {
    local bitmask=0
    for cpu in ${*//,/ }; do
        bitmask=$((bitmask | (1 << cpu)))
    done
    printf '%x' $bitmask
}

set_affinity() {
    local irq_name="$1" affinity="$2" occurrence="${3:-1}"
    awk -v irq_name="$irq_name" -v occurrence="$occurrence" '
        BEGIN{count=0}
        $NF==irq_name {
            if(++count==occurrence){
                sub(/:$/,"",$1)
                print $1
            }
    }' /proc/interrupts | while read -r irq; do
        bitmask=$(cpus_to_bitmask "$affinity")
        echo "$bitmask" > /proc/irq/"$irq"/smp_affinity 2>/dev/null && \
        $enable_log && log_msg "$irq_name" "$affinity" "$occurrence" "$irq"
    done
}

balance_interrupts() {
    logger -t "$PROG" "Optimizing IRQ balancing"

    set_affinity "nss_queue0" "1,2,3" 1  
    set_affinity "nss_queue0" "1,2,3" 2  
    set_affinity "nss_queue1" "0,2,3" 1
    set_affinity "nss_queue2" "0,1,3" 1
    set_affinity "nss_queue3" "0,1,2" 1
    set_affinity "nss_empty_buf_sos" "1,2,3" 1  
    set_affinity "nss_empty_buf_queue" "1,2,3" 1  
    set_affinity "nss_empty_buf_sos" "1,2,3" 2  
    set_affinity "xhci-hcd:usb1" "1,2,3" 1 
    set_affinity "xhci-hcd:usb3" "0,1,2" 1  
}

optimize_rps() {
    NEW_MASK="e"
    
    for netdev in /sys/class/net/*; do
        [ -d "$netdev" ] || continue
        devname=$(basename "$netdev")
        case "$devname" in
            lo|br-*) continue ;;
        esac
        
        for queue in "$netdev"/queues/rx-*; do
            [ -f "$queue/rps_cpus" ] || continue
            echo "$NEW_MASK" > "$queue/rps_cpus" 2>/dev/null && \
            logger -t "$PROG" "RPS set to CPUs 1-3 for $devname"
        done
    done
}

start() {
    config_load nss
    config_get_bool enable_log "general" enable_log 1
    [ "$enable_log" -eq 1 ] && enable_log=true || enable_log=false

    balance_interrupts
    optimize_rps
    
    sysctl -w net.core.netdev_max_backlog=30000 >/dev/null
    sysctl -w net.core.somaxconn=4096 >/dev/null
    sysctl -w net.ipv4.tcp_tw_reuse=1 >/dev/null
    
    echo "$(cpus_to_bitmask "1,2,3")" > /proc/irq/default_smp_affinity 2>/dev/null
}

stop() {
    :
}

restart() {
    stop
    start
}
```

Razifadm, 2025-06-02:
> > > > So from my pov, using your repo, doesn't need any irqbalance, or scrip tweak right? Like, "compile"-flash- voila nss. But cpu0 still spike. Looking forward for adding dhcp interface. As suggested above
> > > 
> > > 
> > > Correct, a lot of my scripts are meant to be hands off approach to getting NSS working across the board (LAN/WAN/WIfi).
> > > If you know the IRQ and affinity I can try getting it merge in.
> > 
> > 
> > i think i found the culprit. for me it was IRQ 43 (xhci usb3), by default affinity is 9 binary(1001) :cpu0 and cpu3. change affinity to mask E binary 1110, cpu1 will spike. hahaha. please sir get in touch here whenever you have the time. thanksss
> 
> **try /etc/init.d/qca-nss-drv**
> 
> ```
> #!/bin/sh /etc/rc.common
> # NSS Driver IRQ Balancing Script - Auto USB IRQ Detection Version
> 
> START=94
> PROG="qca-nss-drv"
> 
> # Get CPU info
> TOTAL_CPUS=$(grep -c ^processor /proc/cpuinfo)
> [ "$TOTAL_CPUS" -lt 4 ] && CPU_LIST="0-$((TOTAL_CPUS-1))" || CPU_LIST="0-$((TOTAL_CPUS-1))"
> 
> CPU_RANGE=""
> for i in $(seq 1 $((TOTAL_CPUS - 1))); do
>     CPU_RANGE="$CPU_RANGE,$i"
> done
> CPU_RANGE="${CPU_RANGE#,}"
> 
> log_msg() {
>     logger -t "$PROG" "Pinning IRQ($4) $1 to CPU $2"
> }
> 
> cpus_to_bitmask() {
>     local bitmask=0
>     for cpu in ${*//,/ }; do
>         bitmask=$((bitmask | (1 << cpu)))
>     done
>     printf '%x' $bitmask
> }
> 
> set_affinity() {
>     local irq_name="$1" affinity="$2" occurrence="${3:-1}"
>     awk -v irq_name="$irq_name" -v occurrence="$occurrence" '
>         BEGIN{count=0}
>         $NF==irq_name {
>             if(++count==occurrence){
>                 sub(/:$/,"",$1)
>                 print $1
>             }
>     }' /proc/interrupts | while read -r irq; do
>         bitmask=$(cpus_to_bitmask "$affinity")
>         echo "$bitmask" > /proc/irq/"$irq"/smp_affinity 2>/dev/null && \
>         $enable_log && log_msg "$irq_name" "$affinity" "$occurrence" "$irq"
>     done
> }
> 
> balance_interrupts() {
>     logger -t "$PROG" "Optimizing IRQ balancing"
>  
>     set_affinity "nss_queue0" "1,2,3" 1
>     set_affinity "nss_queue0" "1,2,3" 2
>     set_affinity "nss_queue1" "0,2,3" 1
>     set_affinity "nss_queue2" "0,1,3" 1
>     set_affinity "nss_queue3" "0,1,2" 1
> 
>     set_affinity "nss_empty_buf_sos" "1,2,3" 1
>     set_affinity "nss_empty_buf_queue" "1,2,3" 1
>     set_affinity "nss_empty_buf_sos" "1,2,3" 2
>  
>     for irq_name in $(awk '{print $NF}' /proc/interrupts | grep -E '^xhci[-_]hcd:usb[0-9]+$'); do
>         set_affinity "$irq_name" "$CPU_RANGE" 1
>     done
> 
>     # ppdu-end interrupts (Wi-Fi)
>     set_affinity "ppdu-end-interrupts-mac1" "1" 1
>     set_affinity "ppdu-end-interrupts-mac3" "2" 1
> }
> 
> optimize_rps() {
>     NEW_MASK="e"  # 0xE = CPUs 1,2,3
> 
>     for netdev in /sys/class/net/*; do
>         [ -d "$netdev" ] || continue
>         devname=$(basename "$netdev")
>         case "$devname" in
>             lo|br-*) continue ;;
>         esac
>         
>         for queue in "$netdev"/queues/rx-*; do
>             [ -f "$queue/rps_cpus" ] || continue
>             echo "$NEW_MASK" > "$queue/rps_cpus" 2>/dev/null && \
>             logger -t "$PROG" "RPS set to CPUs 1-3 for $devname"
>         done
>     done
> }
> 
> start() {
>     config_load nss
>     config_get_bool enable_log "general" enable_log 1
>     [ "$enable_log" -eq 1 ] && enable_log=true || enable_log=false
> 
>     balance_interrupts
>     optimize_rps
>  
>     sysctl -w net.core.netdev_max_backlog=30000 >/dev/null
>     sysctl -w net.core.somaxconn=4096 >/dev/null
>     sysctl -w net.ipv4.tcp_tw_reuse=1 >/dev/null
>  
>     echo "$(cpus_to_bitmask "1,2,3")" > /proc/irq/default_smp_affinity 2>/dev/null
> }
> 
> stop() { :; }
> restart() { stop; start; }
> ```

So you no cpu spike already after this script?  On me IRQ 43, xhci usb3. Already assigne affinity either change cpu or assign to all cpu it still spike. Bit i will try this.

qosmio, 2025-06-02:
>```sh
> for queue in "$netdev"/queues/rx-*; do
>             [ -f "$queue/rps_cpus" ] || continue
>             echo "$NEW_MASK" > "$queue/rps_cpus" 2>/dev/null && \
>             logger -t "$PROG" "RPS set to CPUs 1-3 for $devname"
> done
>```

Please don't do this... This is essentially enabling packet flow steering which conflicts with the way NSS operates with internal ports. They need to be kept on CPU 0. https://github.com/qosmio/openwrt-ipq?tab=readme-ov-file#important-note.

The only devices that should have that set are the radios. Which I already handle as part of the `/etc/init.d/qca-nss-pbuf` init script.

Razifadm, 2025-06-02:
> Please don't do this... This is essentially enabling packet flow steering which conflicts with the way NSS operates with internal ports. They need to be kept on CPU 0. https://github.com/qosmio/openwrt-ipq?tab=readme-ov-file#important-note.

so this is why nothing change still IRQ 43. spike like hellll,,, any thoughts qosmio sir?

qosmio, 2025-06-02:
You should be able to rebuild `qca-nss-drv`. I've updated the init script to account for the additional usb https://github.com/qosmio/nss-packages/commit/2f4750a191614e57fa5699bab1c0c4d66c6924b5

Also, verify you're actually seeing NSS offloading `rmnet_rx`. 

Check with `nss_stats`

Razifadm, 2025-06-02:
> You should be able to rebuild `qca-nss-drv`. I've updated the init script to account for the additional usb [qosmio/nss-packages@2f4750a](https://github.com/qosmio/nss-packages/commit/2f4750a191614e57fa5699bab1c0c4d66c6924b5)
> 
> Also, verify you're actually seeing NSS offloading `rmnet_rx`.
> 
> Check with `nss_stats`

Ive rebuild the qca-nss-drv. now the cpu spike on cpu2. following your recent commit xhci-usb3 pinning to cpu2. https://github.com/qosmio/nss-packages/commit/2f4750a191614e57fa5699bab1c0c4d66c6924b5.

nss_stats

root@Raducksijaa:~# nss_stats

________________________________________________________________________________

<<<<<<<<<<<<<<<<<<<<<<<<<<<<< INTERNAL: IPQ8074 V2 >>>>>>>>>>>>>>>>>>>>>>>>>>>>>
________________________________________________________________________________

________________________________________________________________________________

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< PDEV 0 >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
________________________________________________________________________________

#txrx

        wifili[0]_rx_deliverd                   = 503024          special
        wifili[0]_tx_enqueue                    = 1617669         special
        wifili[0]_tx_dequeue                    = 1617665         special
        wifili[0]_tx_sent_count                 = 1617672         special

#tcl ring

        wifili[0]_tcl_ring_sent  = 1617665         special

#tcl comp

        wifili[0]_tx_desc_free_reaped           = 1653665         special

#reo ring

        wifili[0]_reo_reaped         = 503031          special

#tx sw pool

        wifili[0]_tx_desc_free_src_fw       = 162             special
        wifili[0]_tx_desc_free_completion   = 1617492         special

#tx ext sw pool

#rxdma pool

        wifili[0]_rx_desc_in_use     = 4095            special

#rxdma ring

        wifili[0]_rxdma_buf_replenished     = 543117          special

________________________________________________________________________________

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< PDEV 1 >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
________________________________________________________________________________

#txrx

        wifili[1]_tx_enqueue                    = 168             special
        wifili[1]_tx_dequeue                    = 168             special
        wifili[1]_tx_sent_count                 = 168             special

#tcl ri_sent_count                 = 168             special

#tcl ring
[... 203 lines cut ...]
        drv_tx_buffer_pkt             = 1617863         special
        drv_tx_buffers_cmd            = 340             special
        drv_tx_buffers_reuse          = 340             special
        drv_rx_buffers_empty          = 1617861         special
        drv_rx_buffers_pkt            = 502805          special
        drv_rx_buffers_status_sync    = 296750          special
        drv_tx_skb_simple             = 1617863         special
        drv_rx_skb_simple             = 799555          special
        drv_nss_skb_count             = 14220           special
        drv_tx_buffers_data_queue[0]  = 1617830         special
        drv_tx_buffers_data_queue[1]  = 33              special

________________________________________________________________________________

CPU Utilization:
Note: Averaged over 1 second

Core 0:
Min     Avg     Max
 6%      6%      6%

________________________________________________________________________________
root@Raducksijaa:~# 

where to look rrmnet_rx?

cjayacopra, 2025-06-02:
> root@Raducksijaa:~#
where to look rrmnet_rx?

Did you enabled RMNET in qca-nss-drv?

qosmio, 2025-06-02:
> where to look rrmnet_rx?

```sh
ls /sys/kernel/debug/qca-nss-drv/stats
```

If it's not there, then it's not compiled in the `qca-nss-drv`. Or, if it is there and it's blank that means NSS is not hooking into it.

Razifadm, 2025-06-02:
> > root@Raducksijaa:~#
> > where to look rrmnet_rx?
> 
> Did you enabled RMNET in qca-nss-drv?
> 
> `CONFIG_NSS_DRV_RMNET_ENABLE=y`

I believe so. even kmod-rmnet also picked.

Razifadm, 2025-06-02:
> something like this: `cat /sys/kernel/debug/qca-nss-drv/stats/rmnet_rx`

no device or directories. but for sure i pick both =y. will rebuild with new folder instead. for now pending spike IRQ xhci usb3.

closed 2025-06-03 by qosmio

qosmio, 2025-06-03:
> Hi @qosmio ,
> 
> I know that you have closed this case already. However, it seems that building a firmware using [NSS-12.5-K6.x-wwan)](https://github.com/qosmio/nss-packages/tree/NSS-12.5-K6.x-wwan) is resulting in compilation error. 

Please open an issue in the nss-repo repository with logs and I'll take a look. Thanks.

sipitongyo, 2025-06-04:
OpenWrt IPQ CPU usage is already good. You should know that NSS doesn't run when you use a custom TTL (Time To Live) like 64 or similar,  that's why your CPU usage spikes. If you want to use a custom TTL, use ImmortalWrt or official OpenWrt. Correct me if I'm wrong @qosmio