qosmio/openwrt-ipq #51: Build issues on latest commits

alryaz, 2025-06-18:
Hello!

I've been having build issues after commit `a0ea0a8e6d949067fd937c6fc6fc3bbba364cbcf`, specifically cacheline assertion. I would love to attribute it to my inclination to include `kmod-netdev` package, however I am really unsure about that. 

The complete error looks like:
```
In file included from <command-line>:
In function 'net_dev_struct_check',
    inlined from 'net_dev_init' at net/core/dev.c:12166:2:
././include/linux/compiler_types.h:536:45: error: call to '__compiletime_assert_1674' declared with attribute error: BUILD_BUG_ON failed: offsetof(struct net_device, __cacheline_group_end__net_device_read_tx) - offsetofend(struct net_device, __cacheline_group_begin__net_device_read_tx) > 160
  536 |         _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |                                             ^
././include/linux/compiler_types.h:517:25: note: in definition of macro '__compiletime_assert'
  517 |                         prefix ## suffix();                             \
      |                         ^~~~~~
././include/linux/compiler_types.h:536:9: note: in expansion of macro '_compiletime_assert'
  536 |         _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |         ^~~~~~~~~~~~~~~~~~~
./include/linux/build_bug.h:39:37: note: in expansion of macro 'compiletime_assert'
   39 | #define BUILD_BUG_ON_MSG(cond, msg) compiletime_assert(!(cond), msg)
      |                                     ^~~~~~~~~~~~~~~~~~
./include/linux/build_bug.h:50:9: note: in expansion of macro 'BUILD_BUG_ON_MSG'
   50 |         BUILD_BUG_ON_MSG(condition, "BUILD_BUG_ON failed: " #condition)
      |         ^~~~~~~~~~~~~~~~
./include/linux/cache.h:167:9: note: in expansion of macro 'BUILD_BUG_ON'
  167 |         BUILD_BUG_ON(offsetof(TYPE, __cacheline_group_end__##GROUP) - \
      |         ^~~~~~~~~~~~
net/core/dev.c:12063:9: note: in expansion of macro 'CACHELINE_ASSERT_GROUP_SIZE'
12063 |         CACHELINE_ASSERT_GROUP_SIZE(struct net_device, net_device_read_tx, 160);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
make[9]: *** [scripts/Makefile.build:229: net/core/dev.o] Error 1
make[8]: *** [scripts/Makefile.build:478: net/core] Error 2
make[7]: *** [scripts/Makefile.build:478: net] Error 2
make[6]: *** [/home/runner/work/openwrt-ax3600-builds/openwrt-ax3600-builds/build_repo/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.12.33/Makefile:1945: .] Error 2
make[5]: *** [Makefile:224: __sub-make] Error 2
make[5]: Leaving directory '/home/runner/work/openwrt-ax3600-builds/openwrt-ax3600-builds/build_repo/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.12.33'
make[4]: *** [Makefile:24: /home/runner/work/openwrt-ax3600-builds/openwrt-ax3600-builds/build_repo/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.12.33/.modules] Error 2
make[4]: Leaving directory '/home/runner/work/openwrt-ax3600-builds/openwrt-ax3600-builds/build_repo/target/linux/qualcommax'
make[3]: *** [Makefile:12: compile] Error 2
make[3]: Leaving directory '/home/runner/work/openwrt-ax3600-builds/openwrt-ax3600-builds/build_repo/target/linux'
time: target/linux/compile#3.45#0.44#3.[86](https://github.com/alryaz/openwrt-ax3600-builds/actions/runs/15732341715/job/44336243626#step:15:87)
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/home/runner/work/openwrt-ax3600-builds/openwrt-ax3600-builds/build_repo'
make[1]: *** [target/Makefile:25: /home/runner/work/openwrt-ax3600-builds/openwrt-ax3600-builds/build_repo/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/runner/work/openwrt-ax3600-builds/openwrt-ax3600-builds/build_repo'
make: *** [/home/runner/work/openwrt-ax3600-builds/openwrt-ax3600-builds/build_repo/include/toplevel.mk:233: world] Error 2
```

I am building against `main-nss` repository.

What could start triggering this behavior?

suyuan168, 2025-06-26:
I have also encountered this problem. Looking for help if there is any solution.

alryaz, 2025-06-26:
@suyuan168 could you, please, attach your .config file?
I want to cross-examine, as I will have two failed ones and one successful one.

qosmio, 2025-07-30:
fixed in a478f98a5eff5820c66cb57b1dd165f05db6c7a4

closed 2025-07-30 by qosmio